InfiniTimeOrg/InfiniSim #11: Exiting "Chimes" setting crashes Infinisim

edo0, 2022-03-08:
Steps to reproduce: 
- enter the Settings
- open the Chimes settings item
- click and drag from top to bottom to exit | other times, the behaviour is obtained by just entering the Chimes setting, without clicking and dragging
- InfiniSim crashes

This is what the terminal read  
```
$ ./build/infinisim
initial free_size = 25648
info:  systemtask task started!
info:  Last reset reason : Reset pin
info:  [SpiNorFlash] Manufacturer : 0, Memory type : 0, memory density : 0
info:  [SpiNorFlash] Wakeup
info:  displayapp task started!
info:  DISABLE
info:  DISABLE
free(): invalid next size (fast)
Aborted

```

NeroBurner, 2022-03-08:
can reproduce, backtrace:

```
libc.so.6!__pthread_kill_implementation (Unknown Source:0)
libc.so.6!raise (Unknown Source:0)
libc.so.6!abort (Unknown Source:0)
libc.so.6!__libc_message (Unknown Source:0)
libc.so.6!malloc_printerr (Unknown Source:0)
libc.so.6!unlink_chunk.constprop (Unknown Source:0)
libc.so.6!malloc_consolidate (Unknown Source:0)
libc.so.6!_int_malloc (Unknown Source:0)
libc.so.6!malloc (Unknown Source:0)
libxcb.so.1![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libxcb.so.1!xcb_wait_for_reply64 (Unknown Source:0)
libX11.so.6!_XReply (Unknown Source:0)
libX11.so.6!XkbGetUpdatedMap (Unknown Source:0)
libSDL2-2.0.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
sdl_event_handler(lv_task_t * t) (/home/nero/repos/pinetime/InfiniSim/lv_drivers/display/monitor.c:233)
lv_task_exec(lv_task_t * task) (/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/libs/lvgl/src/lv_misc/lv_task.c:386)
lv_task_handler() (/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/libs/lvgl/src/lv_misc/lv_task.c:134)
Pinetime::Applications::DisplayApp::Refresh(Pinetime::Applications::DisplayApp * const this) (/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/displayapp/DisplayApp.cpp:143)
Pinetime::Applications::DisplayApp::Process(void * instance) (/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/displayapp/DisplayApp.cpp:124)
sdl_function_wrapper(void * instance) (/home/nero/repos/pinetime/InfiniSim/sim/task.cpp:61)
```

NeroBurner, 2022-03-08:
congratulations @edo0 you've found a memory corruption in InfiniTime, that somehow didn't effect the firmware, but broke the simulator (it's an error either way). Thanks!

opened a PR to fix it https://github.com/InfiniTimeOrg/InfiniTime/pull/1026

edo0, 2022-03-08:
Thank you for your work on InfiniSim!!
I have yet to receive my first Pinetime (expected in a few weeks), but I am happy to see I already had some findings just by playing around with the simulator.

JF002, 2022-03-09:
@edo0 Thanks for reporting the issue, I've just merged the fix in InfiniTime :)

closed 2022-03-09 by NeroBurner