InfiniTimeOrg/InfiniSim #13: Suggestions: Simulate the speed of the real Pinetime.

CoderThomasB, 2022-03-10:
On the real Pinetime, there is a little animation for scrolling that doesn't show up on the simulated. I assume it's because the clock speed on my laptop is way faster than the Pinetime. Is it possible to purposely slow down the speed of the code to simulate the speed of the real Pinetime and therefore display the animations?

NeroBurner, 2022-03-10:
It's not the speed, it's just the lack of switching animation :see_no_evil: 

I haven't implemented the screen switching animations and I don't quite know how. As far as I know the pinetime has some kind of hardware support for this switching, but I don't know much more about it.

Would be really awesome if the switching could get implemented

NeroBurner, 2022-03-10:
maybe the screen loading API can help https://docs.lvgl.io/7.11/overview/object.html#load-screens

NeroBurner, 2022-06-13:
Fixed with #37 

[image]

closed 2022-06-13 by NeroBurner