InfiniTimeOrg/InfiniSim #16: Issues with turning the screen off.

Riksu9000, 2022-03-21:
When the screen turns off while InfiniSim is launched with the status window, the screen freezes, instead of turning the screen black and reading "Screen is OFF".

When InfiniSim is launched without the status window, the Interface is entirely black, and when the screen turns off, the "Screen is OFF" text is shown. This text doesn't go away when trying to wake it up.

These issues aren't present on my Pinebook Pro. I have an Nvidia gpu and the issue is present on wayland and xorg, if that means anything.

I thought these issues might be related, so I'm posting them in a single issue.

https://user-images.githubusercontent.com/37774658/159255017-bb7677b5-c25e-4653-b08f-37cc9e995a51.mp4

NeroBurner, 2022-03-21:
could you try with the `settings.dat` file moved or removed?

Riksu9000, 2022-03-21:
I tried removing the file, but the issue persists.

Riksu9000, 2022-03-21:
This issue also happens in builds created by github actions on InfiniTime, which should rule out issues in the build.

NeroBurner, 2022-03-21:
can't reproduce on current InfiniSim `main` branch https://github.com/InfiniTimeOrg/InfiniSim/commit/d6cc748ea1903050b8738d58c03e97b61cee1610 and InfiniTime `develop` branch https://github.com/InfiniTimeOrg/InfiniTime/commit/1c4a56b05bf4fcd89d95a7c04e5a7a22f554f00c

Can you test with those exact commits?

tried with and without the status screen. A right mouse-click always wakes the screen up for me

I'm on Arch Linux, KDE Plasma 5.24.3 on with Wayland compositor and a AMD Radeon RX 580. Don't have a NVidia card to test it

Riksu9000, 2022-03-21:
Tried those commits, no difference. I should clarify that the simulator keeps working even with this issue. Waking it up it refreshes the screen and it keeps working, and even when the screen is entirely black, I think it's still running correctly according to the status messages when right clicking.

I checked that the code execution does enter the part where the screen off objects get created by adding printfs, yet the objects can't be seen on screen. Could they be offscreen somehow?

Lastly I made an interesting discovery. I tried this on another machine with an nvidia card and it worked at first. Then I switched from nouveau drivers to the proprietary ones, and now it has the issue as well!

Riksu9000, 2022-03-22:
Minimizing doesn't help either. I can't say I have never encountered issues, but pretty much everything works "as expected" on the proprietary drivers as far as I can tell. I can't remember having encountered anything like this before.

Da-Viper, 2023-03-21:
This seems to be a nvidia proprietary drivers problem, I disable the nvidia card from the BIOS and it turns off properly but when I enable it the issues persists.

NeroBurner, 2023-05-03:
does this issue still exist or did SDL or NVidia maybe fix the problem?

Da-Viper, 2023-05-03:
> does this issue still exist or did SDL or NVidia maybe fix the problem?

Yes this issue still occurs, I am on 
SDL2 version:  2.26.5
Nvidia Driver version:  530.41.03

NeroBurner, 2023-05-11:
Could you try the original lvgl simulator with NVidia hardware. I'd like to try narrowing down if `InfiniSim` does something wonky, or the lvgl-simulator `InfiniSim` is based on, or SDL2

https://github.com/lvgl/lv_port_pc_eclipse/tree/dev-7.0 (please try the lvgl7 branch `dev7`)