InfiniTimeOrg/InfiniSim #19: libpng errors during compilation on Fedora

Wivik, 2022-04-02:
Hello,

I had some issues during the compilation, here is how I've fixed them : 

```
InfiniSim/main.cpp:64:10: fatal error: libpng16/png.h: No such file or directory
```

Changed `#include <libpng16/png.h>` to `#include <libpng/png.h>` in `main.cpp` : OK but new error

```
InfiniSim/libpng/png.h:330:13: fatal error: pnglibconf.h: No such file or directory
```

Copied scripts/pnglibconf.h.prebuilt to pnglibconf.h

Build and run OK.

I don't know the impact for other systems (I'm not a dev) so I prefer not proposing any changes. If it's ok for you, I can propose a "Troubleshoot" section in the README with these informations.

NeroBurner, 2022-04-02:
That's strange. I specifically added libpng as git submodule to not add another external build dependency and thus should work on all platforms in the same way.

Are you sure you have checked out the libpng submodule?

To have a better understanding could you provide the version of your build tools like CMake and GCC? And maybe also which OS you're running?

Wivik, 2022-04-02:
I've followed the steps in the readme, my clone looks like this : 

```
drwxr-xr-x.  4 seb seb 4.0K Apr  2 23:25 build
drwxr-xr-x.  2 seb seb 4.0K Apr  2 22:37 cmake
-rw-r--r--.  1 seb seb 9.9K Apr  2 22:37 CMakeLists.txt
drwxr-xr-x. 14 seb seb 4.0K Apr  2 22:37 InfiniTime
drwxr-xr-x. 11 seb seb 4.0K Apr  2 22:47 libpng
-rw-r--r--.  1 seb seb  35K Apr  2 22:37 LICENSE
drwxr-xr-x.  8 seb seb 4.0K Apr  2 22:37 lv_drivers
-rw-r--r--.  1 seb seb  13K Apr  2 22:37 lv_drv_conf.h
-rw-r--r--.  1 seb seb  36K Apr  2 22:41 main.cpp
-rw-r--r--.  1 seb seb 3.7K Apr  2 22:37 README.md
drwxr-xr-x.  8 seb seb 4.0K Apr  2 22:37 sim
```

The submodules are present, but libpng is named "libpng", not "libpng16". I think that's the problem for the first point.

```
[core]
	repositoryformatversion = 0
	filemode = true
	bare = false
	logallrefupdates = true
[submodule]
	active = .
[remote "origin"]
	url = https://github.com/InfiniTimeOrg/InfiniSim.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "main"]
	remote = origin
	merge = refs/heads/main
[submodule "InfiniTime"]
	url = https://github.com/InfiniTimeOrg/InfiniTime.git
[submodule "libpng"]
	url = https://github.com/glennrp/libpng.git
[submodule "lv_drivers"]
	url = https://github.com/lvgl/lv_drivers.git
```

I'm on Fedora 34, using `cmake version 3.20.5` and `gcc (GCC) 11.2.1 20220127 (Red Hat 11.2.1-9)`.

NeroBurner, 2022-04-03:
hmmm my `InfiniSim` repo looks just like yours. Especially the `libpng` submodule is the folder `libpng`

```
drwxr-xr-x  2 nero nero  4096 31. Mär 22:04 cmake
-rw-r--r--  1 nero nero 10136  2. Apr 17:52 CMakeLists.txt
drwxr-xr-x 15 nero nero  4096  2. Apr 17:52 InfiniTime
drwxr-xr-x 11 nero nero  4096  8. Mär 21:34 libpng
-rw-r--r--  1 nero nero 35148 31. Mär 22:04 LICENSE
drwxr-xr-x  8 nero nero  4096 16. Feb 21:58 lv_drivers
-rw-r--r--  1 nero nero 12380 31. Mär 22:04 lv_drv_conf.h
-rw-r--r--  1 nero nero 36693  2. Apr 17:52 main.cpp
-rw-r--r--  1 nero nero  3779 31. Mär 22:04 README.md
-rw-r--r--  1 nero nero    40 30. Mär 20:09 settings.dat
```

But you're right, something is "fishy". When I let vscode tell me where `libpng16/png.h` is it opens the system include `/usr/include/libpng16/png.h`

NeroBurner, 2022-04-03:
@Wivik I've created a PR to fix the libpng submodule usage. Thanks for finding this bug!

Could you please check if the solution in https://github.com/InfiniTimeOrg/InfiniSim/pull/20 fixes your problem?

Wivik, 2022-04-03:
@NeroBurner Confirmed, the build is successful at first try with the fix branch !

The two issues are resolved, thanks.

NeroBurner, 2022-04-03:
Thank you for reporting this bug and checking the fix :)

closed 2022-04-03 by NeroBurner