InfiniTimeOrg/InfiniSim #23: Simulator breaks when using paths that start with /

Arsen6331, 2022-04-08:
When using a path starting with `/`, the simulator does not detect the file. I imagine it's trying to look in the root of my computer's filesystem. Instead, if a leading slash exists, it should be removed or a `.` should be added, so that the path is relative to the current directory.

NeroBurner, 2022-04-09:
Absolute paths are trimmed to be paths relative to the working directory.

```cpp
  const char *local_filename = fileName[0]=='/' ? &fileName[1] : fileName;
```

See https://github.com/InfiniTimeOrg/InfiniSim/blob/main/sim/components/fs/FS.cpp#L63

Please provide more more info what you are doing and what isn't working

Arsen6331, 2022-04-09:
It seems to be specifically the Stat() function that isn't working. I should've been more specific.

NeroBurner, 2022-04-09:
THAT is totally plausible! It is new code and I didn't watch out for the relative path thingy. 

Please update the description and title of the issue to reflect that the stat function is faulty.

And a PR is very welcome :)

closed 2022-04-24 by NeroBurner