InfiniTimeOrg/InfiniSim #25: Problems building on OS X

tnlogy, 2022-04-21:
Have anyone managed to get InfiniSim running on OS X? I've made a small attempt, but haven't had much success. I did get it to show the start screen, but no events are working (maybe due to that I had to rewrite some code).

Summary of making it compile:
Change to using https://github.com/trenki2/SDL2Test/blob/master/cmake/FindSDL2.cmake to find SDL. (Installed with the downloadable dmg-file with development libraries from SDL homepage https://www.libsdl.org/download-2.0.php)

Then I had to change some code in the Alarmcontroller.cpp:
`
  auto now = std::chrono::system_clock::now();
  alarmTime = now;
  time_t ttAlarmTime = std::chrono::system_clock::to_time_t(now);
`

Then I had to remove the call:
`
  // SDL_CreateThread(tick_thread, "tick", NULL);
`

And modify the while loop to force the UI to be visible:
`
  SDL_Event e;
  bool quit = false;
  while(!quit) {
      SDL_PollEvent(&e);
      if (e.type == SDL_QUIT){
                  quit = true;
      }
      // SDL_Delay(LV_DISP_DEF_REFR_PERIOD);
      // lv_tick_inc(LV_DISP_DEF_REFR_PERIOD); // Crashed when called here or in tick_thread.
      fw.handle_keys(); // key event polling
      fw.handle_touch_and_button();
      fw.refresh();
      usleep(LV_DISP_DEF_REFR_PERIOD * 1000);
  }
`

If I run it with unmodified code it crashes with: `Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'`

Maybe this can be of help if someone else tries to make it work on a Mac?

adrienluitot, 2023-03-19:
I couldn't build on Mac OS Ventura (M2 Apple silicon) neither.

I had the error : 
```
DateTimeController.cpp:72:66: error: no viable conversion from 'time_point<[...], duration<[...], ratio<[...], 1000000000>>>' to 'const time_point<[...], duration<[...], ratio<[...], 1000000>>>'
  std::time_t currentTime = std::chrono::system_clock::to_time_t(currentDateTime);
```

Thus, I made a change on `DateTimeController.cpp` on line 72.
I added a cast like this : 
```cpp
std::time_t currentTime = std::chrono::system_clock::to_time_t(
    std::chrono::time_point_cast<std::chrono::system_clock::duration>(currentDateTime)
);
```
---
Then, to be able to run the simulation, I also did the changes to `main.cpp` (As tnlogy said above) to fix this error :
`Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'`

But I pretty sure we can find a better fix.

PS: I haven't investigate yet and I don't know what is the given infinitime firmware, but nothing seems to work. I can't quit (without a kill), can't touch anything...

ianmlunaq, 2024-01-25:
I had the same error as @adrienluitot when trying to build on macOS Ventura with an M1 Apple chip. I replaced line 72 in `DateTimeController.cpp` with:
```c++
std::time_t currentTime = std::chrono::time_point_cast<std::chrono::microseconds>(currentDateTime).time_since_epoch().count();
```
That seemed to work and I got to 100% when building, but then I got this error:
```
[100%] Linking CXX executable infinisim
Undefined symbols for architecture arm64:
  "__lv_anim_ll", referenced from:
      __lv_anim_core_init in libsim-base.a(lv_anim.c.o)
      _anim_task in libsim-base.a(lv_anim.c.o)
      _anim_mark_list_change in libsim-base.a(lv_anim.c.o)
      _lv_anim_start in libsim-base.a(lv_anim.c.o)
      _lv_anim_del in libsim-base.a(lv_anim.c.o)
      _lv_anim_del_all in libsim-base.a(lv_anim.c.o)
      _lv_anim_get in libsim-base.a(lv_anim.c.o)
      ...
  "__lv_disp_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _lv_obj_get_disp in libsim-base.a(lv_obj.c.o)
      _lv_disp_drv_register in libsim-base.a(lv_hal_disp.c.o)
      _lv_disp_remove in libsim-base.a(lv_hal_disp.c.o)
      _lv_disp_get_next in libsim-base.a(lv_hal_disp.c.o)
  "__lv_draw_mask_list", referenced from:
      _lv_draw_mask_add in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_apply in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_remove_id in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_remove_custom in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_get_cnt in libsim-base.a(lv_draw_mask.c.o)
  "__lv_drv_ll", referenced from:
      __lv_fs_init in libsim-base.a(lv_fs.c.o)
      _lv_fs_get_drv in libsim-base.a(lv_fs.c.o)
      _lv_fs_drv_register in libsim-base.a(lv_fs.c.o)
      _lv_fs_get_letters in libsim-base.a(lv_fs.c.o)
  "__lv_font_decompr_buf", referenced from:
      _lv_font_get_bitmap_fmt_txt in libsim-base.a(lv_font_fmt_txt.c.o)
      __lv_font_clean_up_fmt_txt in libsim-base.a(lv_font_fmt_txt.c.o)
  "__lv_img_cache_array", referenced from:
      __lv_img_cache_open in libsim-base.a(lv_img_cache.c.o)
      _lv_img_cache_set_size in libsim-base.a(lv_img_cache.c.o)
      _lv_img_cache_invalidate_src in libsim-base.a(lv_img_cache.c.o)
  "__lv_img_defoder_ll", referenced from:
      __lv_img_decoder_init in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_create in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_get_info in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_open in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_delete in libsim-base.a(lv_img_decoder.c.o)
  "__lv_indev_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _lv_indev_drv_register in libsim-base.a(lv_hal_indev.c.o)
      _lv_indev_get_next in libsim-base.a(lv_hal_indev.c.o)
  "__lv_mem_buf", referenced from:
      __lv_mem_buf_get in libsim-base.a(lv_mem.c.o)
      __lv_mem_buf_release in libsim-base.a(lv_mem.c.o)
      __lv_mem_buf_free_all in libsim-base.a(lv_mem.c.o)
     (maybe you meant: __lv_mem_buf_release, __lv_mem_buf_get , __lv_mem_buf_free_all )
  "__lv_obj_style_trans_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _trans_del in libsim-base.a(lv_obj.c.o)
      _trans_create in libsim-base.a(lv_obj.c.o)
      _trans_anim_ready_cb in libsim-base.a(lv_obj.c.o)
      _lv_obj_finish_transitions in libsim-base.a(lv_obj.c.o)
  "__lv_task_act", referenced from:
      _lv_task_handler in libsim-base.a(lv_task.c.o)
      _lv_task_del in libsim-base.a(lv_task.c.o)
  "__lv_task_ll", referenced from:
      __lv_task_core_init in libsim-base.a(lv_task.c.o)
      _lv_task_handler in libsim-base.a(lv_task.c.o)
      _lv_task_create in libsim-base.a(lv_task.c.o)
      _lv_task_del in libsim-base.a(lv_task.c.o)
      _lv_task_set_prio in libsim-base.a(lv_task.c.o)
      _lv_task_get_next in libsim-base.a(lv_task.c.o)
  "__lv_theme_empty_styles", referenced from:
      _lv_theme_empty_init in libsim-base.a(lv_theme_empty.c.o)
ld: symbol(s) not found for architecture arm64
```
This seems to me like more work has to be done get it to work on ARM chip devices.

BluBloos, 2025-02-23:
I'm having issues as well. I had to make the same change to DateTimeController.cpp as @adrienluitot. I've had to do the same SDL2 stuff.

However, I am running into a different issue. I cannot even get the project to build. I get linker errors:

```
Undefined symbols for architecture x86_64:
  "__lv_anim_ll", referenced from:
      __lv_anim_core_init in libsim-base.a(lv_anim.c.o)
      _anim_task in libsim-base.a(lv_anim.c.o)
      _anim_mark_list_change in libsim-base.a(lv_anim.c.o)
      _lv_anim_start in libsim-base.a(lv_anim.c.o)
      _lv_anim_del in libsim-base.a(lv_anim.c.o)
      _lv_anim_del_all in libsim-base.a(lv_anim.c.o)
      _lv_anim_get in libsim-base.a(lv_anim.c.o)
      ...
  "__lv_disp_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _lv_obj_get_disp in libsim-base.a(lv_obj.c.o)
      _lv_disp_drv_register in libsim-base.a(lv_hal_disp.c.o)
      _lv_disp_remove in libsim-base.a(lv_hal_disp.c.o)
      _lv_disp_get_next in libsim-base.a(lv_hal_disp.c.o)
  "__lv_draw_mask_list", referenced from:
      _lv_draw_mask_add in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_apply in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_remove_id in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_remove_custom in libsim-base.a(lv_draw_mask.c.o)
      _lv_draw_mask_get_cnt in libsim-base.a(lv_draw_mask.c.o)
  "__lv_drv_ll", referenced from:
      __lv_fs_init in libsim-base.a(lv_fs.c.o)
      _lv_fs_get_drv in libsim-base.a(lv_fs.c.o)
      _lv_fs_drv_register in libsim-base.a(lv_fs.c.o)
      _lv_fs_get_letters in libsim-base.a(lv_fs.c.o)
  "__lv_font_decompr_buf", referenced from:
      _lv_font_get_bitmap_fmt_txt in libsim-base.a(lv_font_fmt_txt.c.o)
      __lv_font_clean_up_fmt_txt in libsim-base.a(lv_font_fmt_txt.c.o)
  "__lv_img_cache_array", referenced from:
      __lv_img_cache_open in libsim-base.a(lv_img_cache.c.o)
      _lv_img_cache_set_size in libsim-base.a(lv_img_cache.c.o)
      _lv_img_cache_invalidate_src in libsim-base.a(lv_img_cache.c.o)
  "__lv_img_defoder_ll", referenced from:
      __lv_img_decoder_init in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_create in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_get_info in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_open in libsim-base.a(lv_img_decoder.c.o)
      _lv_img_decoder_delete in libsim-base.a(lv_img_decoder.c.o)
  "__lv_indev_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _lv_indev_drv_register in libsim-base.a(lv_hal_indev.c.o)
      _lv_indev_get_next in libsim-base.a(lv_hal_indev.c.o)
  "__lv_mem_buf", referenced from:
      __lv_mem_buf_get in libsim-base.a(lv_mem.c.o)
      __lv_mem_buf_release in libsim-base.a(lv_mem.c.o)
      __lv_mem_buf_free_all in libsim-base.a(lv_mem.c.o)
     (maybe you meant: __lv_mem_buf_release, __lv_mem_buf_get , __lv_mem_buf_free_all )
  "__lv_obj_style_trans_ll", referenced from:
      _lv_init in libsim-base.a(lv_obj.c.o)
      _trans_del in libsim-base.a(lv_obj.c.o)
      _trans_create in libsim-base.a(lv_obj.c.o)
      _trans_anim_ready_cb in libsim-base.a(lv_obj.c.o)
      _lv_obj_finish_transitions in libsim-base.a(lv_obj.c.o)
  "__lv_task_act", referenced from:
      _lv_task_handler in libsim-base.a(lv_task.c.o)
      _lv_task_del in libsim-base.a(lv_task.c.o)
  "__lv_task_ll", referenced from:
      __lv_task_core_init in libsim-base.a(lv_task.c.o)
      _lv_task_handler in libsim-base.a(lv_task.c.o)
      _lv_task_create in libsim-base.a(lv_task.c.o)
      _lv_task_del in libsim-base.a(lv_task.c.o)
      _lv_task_set_prio in libsim-base.a(lv_task.c.o)
      _lv_task_get_next in libsim-base.a(lv_task.c.o)
  "__lv_theme_empty_styles", referenced from:
      _lv_theme_empty_init in libsim-base.a(lv_theme_empty.c.o)
ld: symbol(s) not found for architecture x86_64
```

Running on macOS 12.7.4 intel silicon.