InfiniTimeOrg/InfiniSim #26: AUR package not working in Manjaro

boschkundendienst, 2022-04-23:
Hi, I tried to install InfiniSim from the AUR on Manjaro and it exits with the following error (while manually builiding based on your documentation works):

```bash
pamac install infinisimi-git
...
...
-- Configuring done
CMake Error at CMakeLists.txt:217 (add_library):
  Cannot find source file:

    /tmp/trizen-pcfreak/infinisim-git/src/infinisim-git/InfiniTime/src/libs/QCBOR/src/qcbor_decode.c

CMake Error at CMakeLists.txt:217 (add_library):
  No SOURCES given to target: QCBOR

CMake Generate step failed.  Build files cannot be regenerated correctly.
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...
:: Unable to build infinisim-git - makepkg exited with code: 4
```

NeroBurner, 2022-04-24:
with https://github.com/InfiniTimeOrg/InfiniSim/pull/22 I've added QCBOR support, but didn't update the aur PKGBUILD. Thanks for the ping

NeroBurner, 2022-04-24:
Package should be fixed now https://aur.archlinux.org/cgit/aur.git/commit/?h=infinisim-git&id=5959c7f6d6ac69434d9ef8547d0bbcda2606d4e4

boschkundendienst, 2022-04-26:
Thanks NeroBurner for the quick response. Works like a charm now.

closed 2022-04-26 by boschkundendienst