InfiniTimeOrg/InfiniSim #27: How do I edit the operating system without breaking the simulator

joseph58tech, 2022-04-24:
I was trying to edit the pinetime folder in the simulator so I can add the infinineat watch face, but when I make changes to the folder it give me the "core dumped" error.

I'm using the InfiniSim on a chromebook if that helps

NeroBurner, 2022-04-24:
could you please provide some minimal commands to reproduce your workflow locally for me? And maybe the error message you're getting, and when you're getting the error message.

Does the unmodified `InfiniSim` + `InfiniTime` folder build and run successfully on your machine?

I think I had the InfiNeat Watchface running locally once in the simulator. Hope we'll get it to run on your machine too

NeroBurner, 2022-08-07:
closing due to inactivity and I have no chromebook to reproduce the issue

closed 2022-08-07 by NeroBurner