InfiniTimeOrg/InfiniSim #32: Font related build issues

faxe1008, 2022-05-14:
Hi,

I tried building this with the following source revisions:
InfiniSim: `d9d729d40f0af12e7c46c917dd0082f33fc4e466`
InfiniTime: `4cb07bac5342fd583e23e1643c2e95f649dc9c59`

In both repositories all submodules are initialized and updated. In invoked the build like this:
```
cmake -S . -B build -DInfiniTime_DIR=/home/faxe/priv/InfiniTime
cmake --build build -j4
```

And get the following output:
```
Consolidate compiler generated dependencies of target QCBOR
[  3%] Built target QCBOR
[  5%] Built target genfiles
Consolidate compiler generated dependencies of target png_static
[ 14%] Built target png_static
Consolidate compiler generated dependencies of target infinisim
[ 14%] Building CXX object CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Clock.cpp.o
[ 15%] Building CXX object CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp.o
[ 15%] Building CXX object CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/DisplayApp.cpp.o
[ 16%] Linking CXX executable infinisim
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp.o: warning: relocation against `jetbrains_mono_extrabold_compressed' in read-only section `.text'
/usr/bin/ld: CMakeFiles/infinisim.dir/main.cpp.o: in function `hal_init()':
main.cpp:(.text+0x927): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: main.cpp:(.text+0x92f): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: main.cpp:(.text+0x936): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: main.cpp:(.text+0x93d): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/libs/lvgl/src/lv_core/lv_obj.c.o: in function `lv_init':
lv_obj.c:(.text+0x12af): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/libs/lvgl/src/lv_core/lv_obj.c.o:lv_obj.c:(.text+0x12b7): more undefined references to `jetbrains_mono_bold_20' follow
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Alarm.cpp.o: in function `Pinetime::Applications::Screens::Alarm::Alarm(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::AlarmController&, Pinetime::Controllers::Settings&, Pinetime::System::SystemTask&)':
Alarm.cpp:(.text+0x200): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: Alarm.cpp:(.text+0x2e1): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/BatteryInfo.cpp.o: in function `Pinetime::Applications::Screens::BatteryInfo::BatteryInfo(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::Battery&)':
BatteryInfo.cpp:(.text+0x42b): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/FlashLight.cpp.o: in function `Pinetime::Applications::Screens::FlashLight::FlashLight(Pinetime::Applications::DisplayApp*, Pinetime::System::SystemTask&, Pinetime::Controllers::BrightnessController&)':
FlashLight.cpp:(.text+0x21c): undefined reference to `lv_font_sys_48'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/HeartRate.cpp.o: in function `Pinetime::Applications::Screens::HeartRate::HeartRate(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::HeartRateController&, Pinetime::System::SystemTask&)':
HeartRate.cpp:(.text+0x1a5): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Metronome.cpp.o: in function `Pinetime::Applications::Screens::Metronome::Metronome(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::MotorController&, Pinetime::System::SystemTask&)':
Metronome.cpp:(.text+0x39c): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: Metronome.cpp:(.text+0x3d9): undefined reference to `jetbrains_mono_bold_20'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Navigation.cpp.o: in function `Pinetime::Applications::Screens::Navigation::Navigation(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::NavigationService&)':
Navigation.cpp:(.text+0x286): undefined reference to `lv_font_navi_80'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Paddle.cpp.o: in function `Pinetime::Applications::Screens::Paddle::Paddle(Pinetime::Applications::DisplayApp*, Pinetime::Components::LittleVgl&)':
Paddle.cpp:(.text+0x345): undefined reference to `jetbrains_mono_42'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/PassKey.cpp.o: in function `Pinetime::Applications::Screens::PassKey::PassKey(Pinetime::Applications::DisplayApp*, unsigned int)':
PassKey.cpp:(.text+0x1ae): undefined reference to `jetbrains_mono_42'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Steps.cpp.o: in function `Pinetime::Applications::Screens::Steps::Steps(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::MotionController&, Pinetime::Controllers::Settings&)':
Steps.cpp:(.text+0x471): undefined reference to `jetbrains_mono_42'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/StopWatch.cpp.o: in function `Pinetime::Applications::Screens::StopWatch::StopWatch(Pinetime::Applications::DisplayApp*, Pinetime::System::SystemTask&)':
StopWatch.cpp:(.text+0x29a): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/Timer.cpp.o: in function `Pinetime::Applications::Screens::Timer::Timer(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::TimerController&)':
Timer.cpp:(.text+0x449): undefined reference to `jetbrains_mono_76'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/WatchFaceDigital.cpp.o: in function `Pinetime::Applications::Screens::WatchFaceDigital::WatchFaceDigital(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::DateTime&, Pinetime::Controllers::Battery&, Pinetime::Controllers::Ble&, Pinetime::Controllers::NotificationManager&, Pinetime::Controllers::Settings&, Pinetime::Controllers::HeartRateController&, Pinetime::Controllers::MotionController&)':
WatchFaceDigital.cpp:(.text+0x5fd): undefined reference to `jetbrains_mono_extrabold_compressed'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/WatchFacePineTimeStyle.cpp.o: in function `Pinetime::Applications::Screens::WatchFacePineTimeStyle::WatchFacePineTimeStyle(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::DateTime&, Pinetime::Controllers::Battery&, Pinetime::Controllers::Ble&, Pinetime::Controllers::NotificationManager&, Pinetime::Controllers::Settings&, Pinetime::Controllers::MotionController&)':
WatchFacePineTimeStyle.cpp:(.text+0x83a): undefined reference to `open_sans_light'
/usr/bin/ld: WatchFacePineTimeStyle.cpp:(.text+0x8fa): undefined reference to `open_sans_light'
/usr/bin/ld: WatchFacePineTimeStyle.cpp:(.text+0x1f66): undefined reference to `lv_font_sys_48'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/settings/QuickSettings.cpp.o: in function `Pinetime::Applications::Screens::QuickSettings::QuickSettings(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::Battery&, Pinetime::Controllers::DateTime&, Pinetime::Controllers::BrightnessController&, Pinetime::Controllers::MotorController&, Pinetime::Controllers::Settings&)':
QuickSettings.cpp:(.text+0x38d): undefined reference to `lv_font_sys_48'
/usr/bin/ld: QuickSettings.cpp:(.text+0x488): undefined reference to `lv_font_sys_48'
/usr/bin/ld: QuickSettings.cpp:(.text+0x5d3): undefined reference to `lv_font_sys_48'
/usr/bin/ld: QuickSettings.cpp:(.text+0x721): undefined reference to `lv_font_sys_48'
/usr/bin/ld: CMakeFiles/infinisim.dir/home/faxe/priv/InfiniTime/src/displayapp/screens/settings/SettingSetTime.cpp.o: in function `Pinetime::Applications::Screens::SettingSetTime::SettingSetTime(Pinetime::Applications::DisplayApp*, Pinetime::Controllers::DateTime&, Pinetime::Controllers::Settings&)':
SettingSetTime.cpp:(.text+0x26e): undefined reference to `jetbrains_mono_42'
/usr/bin/ld: SettingSetTime.cpp:(.text+0x31c): undefined reference to `jetbrains_mono_42'
[... cut ...] me::Controllers::Settings&)':
SettingSteps.cpp:(.text+0x559): undefined reference to `jetbrains_mono_42'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/infinisim.dir/build.make:3782: infinisim] Error 1
make[1]: *** [CMakeFiles/Makefile2:123: CMakeFiles/infinisim.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

There seem to be font related issues. The InfiniTime repositories however builds just fine within docker.

JF002, 2022-05-14:
This is probably caused by [this PR](https://github.com/InfiniTimeOrg/InfiniTime/pull/1097) that was recently merged in InfiniTime : the fonts are now generated at build time. 
InfiniSim needs to integrate these changes. We are currently working on it with [this PR](https://github.com/InfiniTimeOrg/InfiniSim/pull/29) and [this PR](https://github.com/InfiniTimeOrg/InfiniTime/pull/1131).

NeroBurner, 2022-05-16:
with https://github.com/InfiniTimeOrg/InfiniSim/pull/29 and https://github.com/InfiniTimeOrg/InfiniTime/pull/1131 merged this issue should be resolved

closed 2022-05-16 by NeroBurner