InfiniTimeOrg/InfiniSim #39: Add motionController to sim controls

w4tsn, 2022-06-19:
I'd like to control the motion to trigger some shake-based events. For now it would suffice to simulate the accumulated speed in any direction to sim the shake motion calibrated through the shake calibration settings app.

From what I understand of the code, there has to be at least a button assignment in `main.cpp` and probably one can manipulate the `motionController` there. I have yet to play around with that.

My use-case is a Dice app, that I'd like to control with the shake motion. I'm already done for the most part of that app, but want a way to sim this in InfiniSim. Any help / additional info on what parts of the code I have to touch in InfiniSim would help me a lot :)

NeroBurner, 2022-06-19:
For the keyboard interaction you need to update main.CPP like in the screenshot PR https://github.com/InfiniTimeOrg/InfiniSim/pull/10/files

For the simulated motion you'll need to modify the stubbed motioncontroller file, maybe add new function or just implement existing ones. It depends on what you need

w4tsn, 2022-06-21:
Thanks for the link to the PR, that's a good idea / source.

I'm going for generalizing the concept of "Shake" for App interaction and want to start off by using the "Shake gesture" to control the Dice app. I'm currently just taking the "shake calibration for wake on shake" value and multiply that by a fixed amount as the threshold for the "wake on shake" function of the motionController. So to test this in the Sim, I'd have to apply some movement to the motionController (maybe button-press-per-time based) in order to trigger the "wake on shake" function inside this controller. Later on I'd be nice to just trigger the "shake gesture" by button press