InfiniTimeOrg/InfiniSim #41: The default random source seed is different from the nRF52

w4tsn, 2022-06-21:
I'm trying to sim a random generator but found that the C++11 random generator's default random seed source seems to be different in the sim than in the watch. To be more precise, the default random source yields expected results in the sim while it does not work in the watch which is a pity. See the following source code for a demonstration. It will produce random result in the sim while the result will always stay the same on the hardware.

```
uint8_t grade = 20
std::default_random_engine generator {std::random_device()()};
std::uniform_int_distribution<uint8_t> distribution(1, grade);
auto dice = std::bind(distribution, generator);
uint8_t result = dice();
```

I just want to raise this discussion and talk a bit about it. We can close the issue soon afterwards since I don't expect this to be solvable actually. The random source of the watch is probably non-existent / not configured while it is in the sim? Don't know if one can / want to do anything about that.

If we can't get some consistency in the way random works between sim and hardware, any idea on how one would initialize the generator? I was thinking about the motion sensor to be a source of the random seed, maybe combined with time?

NeroBurner, 2022-06-22:
I've tried the C function `rand()` from `#include <cstdlib>` in `Paddle.h` and it seems to create randomness even on the watch

https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/displayapp/screens/Paddle.cpp#L55

w4tsn, 2022-06-23:
Good to know! I'm wondering how that works :thinking:

As `rand()` is somewhat "deprecated" (not really but `random` seems to be preferred in C++11) I've fiddled around with this a bit more and found a working solution using the motionSensor, which I already use anyway in the app, as random seed source:

```
std::seed_seq seed {static_cast<uint16_t>(std::random_device()() % 65535), static_cast<uint16_t>((motionController.X() + motionController.Y() + motionController.Z()) % 32767)};
std::default_random_engine generator {seed};
std::uniform_int_distribution<uint8_t> distribution(1, grade);
auto dice = std::bind(distribution, generator);
```

So far that seems to work nicely. I've added a modulo there to prevent overflow in the `uint16_t`, but I really don't know if that's necessary or not :shrug:

closed 2022-06-23 by w4tsn