InfiniTimeOrg/InfiniSim #47: Trouble building after installing lv_font_conv

toastom, 2022-08-04:
I'm having trouble building the project after the font generation change. I've installed lv_font_conv and the rest of the project dependencies as stated in the README, but when I try to build the project using `cmake --build build -j4`, the build fails when it tries to generate `open_sans_light.c`. It seems to successfully build the other fonts, but for some reason this one is missing.

I run the build command `cmake --build build -j4` and this is the output:
```
toastom@toastom-LM-Laptop:~/GitProjects/InfiniSim$ cmake --build build -j4
[  0%] Generating jetbrains_mono_42.c
[  4%] Built target genfiles
[  5%] Built target QCBOR
[  7%] Generating jetbrains_mono_76.c
[  8%] Generating lv_font_sys_48.c
[  8%] Generating open_sans_light.c
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/convert.js:25:35)
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/convert.js:25:35)
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/convert.js:25:35)
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/home/toastom/GitProjects/InfiniSim/node_modules/lv_font_conv/lib/convert.js:25:35)
Traceback (most recent call last):
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/toastom/GitProjects/InfiniSim/node_modules/.bin/lv_font_conv', '--size', '48', '--output', 'lv_font_sys_48.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/icons_sys_48.ttf', '--range', '0xe902, 0xe904-0xe907, 0xe90b-0xe90c']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_lv_font_sys_48.dir/build.make:61: fonts/lv_font_sys_48.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:396: fonts/CMakeFiles/infinitime_fonts_lv_font_sys_48.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/toastom/GitProjects/InfiniSim/node_modules/.bin/lv_font_conv', '--size', '76', '--output', 'jetbrains_mono_76.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/JetBrainsMono-Light.ttf', '--range', '0x25, 0x2D, 0x2F, 0x30-0x3a']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_76.dir/build.make:61: fonts/jetbrains_mono_76.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:336: fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_76.dir/all] Error 2
Traceback (most recent call last):
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/toastom/GitProjects/InfiniSim/node_modules/.bin/lv_font_conv', '--size', '150', '--output', 'open_sans_light.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/open_sans_light.ttf', '--symbols', '0123456789']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/toastom/GitProjects/InfiniSim/node_modules/.bin/lv_font_conv', '--size', '42', '--output', 'jetbrains_mono_42.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/toastom/GitProjects/InfiniSim/InfiniTime/src/displayapp/fonts/JetBrainsMono-Regular.ttf', '--range', '0x25, 0x2b, 0x2d, 0x30-0x3a']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_open_sans_light.dir/build.make:61: fonts/open_sans_light.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:228: fonts/CMakeFiles/infinitime_fonts_open_sans_light.dir/all] Error 2
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_42.dir/build.make:61: fonts/jetbrains_mono_42.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:201: fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_42.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
toastom@toastom-LM-Laptop:~/GitProjects/InfiniSim$ 
```

NeroBurner, 2022-08-04:
I suspect that the used nodejs version on your distro is too old for the font generation script.

Please try to run the script directly and see what the output tells you

/home/toastom/GitProjects/InfiniSim/node_modules/.bin/lv_font_conv
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

toastom, 2022-08-04:
Running the script without arguments just returns the help screen.
After running with the --version option, it shows version 1.5.2.
`toastom@toastom-LM-Laptop:~/GitProjects/InfiniSim/node_modules/.bin$ ./lv_font_conv --version`
This is strange. This laptop is Linux Mint 20.3. I have another device running Pop OS and that just built the latest InfiniSim without any issue. As far as I understand, they are both based on Ubuntu so they should have the same version of nodejs from the Ubuntu repos.

I did also check the node.js versions.
The successful Pop OS node version
```
node -v
v12.22.9
```

The problem Linux Mint 20.3 node version
```
node -v
v10.19.0
```
So there does seem to be a mismatch here. After work today I will try to manually update my node.js on the problem device and see if that fixes the issue. Thanks.

toastom, 2022-08-05:
Yes! This was the issue. I manually updated my node.js version to the same as my Pop installation using [this guide](https://www.hostingadvice.com/how-to/update-node-js-latest-version/) and the project builds! I will make a PR to document this in the README, if that's okay. Apparently at least version 12 works. Not sure if 11 does, but maybe someone else can verify.

toastom, 2022-08-06:
I think we can close this issue now as it wasn't really an issue with InfiniSim, just some incorrect dependency version.

closed 2022-08-06 by toastom