InfiniTimeOrg/InfiniSim #53: Clarify instructions for installation of lv_font_conv

koreiot, 2022-08-27:
Incredible project.  My sincere thanks to the author and contributors.  I found one item in the README.md that drove me in circles for a few.  Perhaps you can add some additional clarification for others like me, who will likely figure it out, but with a bit of sleuthing... 

The README.md instruction for installing the lv_font_conv modules via npm were a bit vague. It said "Then install the lv_font_conv executable to the source directory (will be installed at node_modules/.bin/lv_font_conv)," but that relative reference to "source" is vague. And, since the sequence of events had this installed before cloning the InfiniSim project, I assumed that source must be somewhere other than within the yet-to-be-cloned InfiniSim project. So, I installed under $HOME (failed to build), then under the InfiniSim directory, to no avail again (both attempts resulted in errors during build due to lv_font_conv not being found).

Then I took a look at the lv_font_conv github page and noticed the use of a global install (-g).  I had been strictly following the README.md instructions which called for a local 'npm install lv_font_conv@1.5.2' installation.  The global npm flag for this module worked, my build completed (related files were found by the build chain), and I am now able to execute the binary (./build/infinisim) and starting to learn how to use this most excellent simulator.

My ask then is to perhaps revisit the README.md and see if you can provide more clarification on exactly where and how to install the lv_font_conv package.  

Thank you!

NeroBurner, 2022-08-29:
You're right, installing `lv_font_conv` is in the README before we get the sources.

Would it be enough in your opinion to move the "Get Soruces" section before the "Build dependencies" section?

RazRosman, 2022-09-03:
Hi,
I'm struggling with the same issue and still didn't mange to build infinisim.
The first build works 
but the second one doesn't and it ends up with an error that lv_font_conv can't be found.

I would love some assistant.

Thanks,
Raz

NeroBurner, 2022-09-03:
Please provide OS, compiler, logs and the used commands for me to reproduce/understand the error. How did you install lv_font_conv?
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

closed 2022-09-05 by NeroBurner