InfiniTimeOrg/InfiniSim #57: Build error when Linking CXX executable infinisim

RazRosman, 2022-09-03:
Hi,

I would love your assistance,

I'm getting the following error when building infiniSim:

[ 68%] Linking CXX executable infinisim
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `command_cp(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool)':
littlefs-do-main.cpp:(.text+0x23fc): undefined reference to `std::filesystem::file_size(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x3c): undefined reference to `std::filesystem::__cxx11::path::has_filename() const'
/usr/bin/ld: littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0xa0): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::is_absolute() const':
littlefs-do-main.cpp:(.text._ZNKSt10filesystem7__cxx114path11is_absoluteEv[_ZNKSt10filesystem7__cxx114path11is_absoluteEv]+0x14): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::exists(std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem6existsERKNS_7__cxx114pathE[_ZNSt10filesystem6existsERKNS_7__cxx114pathE]+0x1c): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::is_directory(std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem12is_directoryERKNS_7__cxx114pathE[_ZNSt10filesystem12is_directoryERKNS_7__cxx114pathE]+0x1c): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::filesystem::__cxx11::path>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::filesystem::__cxx11::path::format)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE[_ZNSt10filesystem7__cxx114pathC5INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE]+0x60): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/littlefs-do.dir/build.make:163: littlefs-do] Error 1
make[1]: *** [CMakeFiles/Makefile2:238: CMakeFiles/littlefs-do.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/infinisim.dir/sim/drivers/SpiNorFlash.cpp.o: in function `std::filesystem::exists(std::filesystem::__cxx11::path const&)':
SpiNorFlash.cpp:(.text._ZNSt10filesystem6existsERKNS_7__cxx114pathE[_ZNSt10filesystem6existsERKNS_7__cxx114pathE]+0x1c): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: CMakeFiles/infinisim.dir/sim/drivers/SpiNorFlash.cpp.o: in function `std::filesystem::__cxx11::path::path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::filesystem::__cxx11::path>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::filesystem::__cxx11::path::format)':
SpiNorFlash.cpp:(.text._ZNSt10filesystem7__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE[_ZNSt10filesystem7__cxx114pathC5INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES1_EERKT_NS1_6formatE]+0x60): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/infinisim.dir/build.make:1427: infinisim] Error 1
make[1]: *** [CMakeFiles/Makefile2:155: CMakeFiles/infinisim.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

NeroBurner, 2022-09-03:
What's your OS and compiler? I suspect GCC is too old to have a std::filesystem implementation

NeroBurner, 2022-09-03:
I'll probably will need to add a cmake option WITH_LITTLEFS_DO or similar, to at least make the simulator useable. If you have other ideas to solve this I'm all ears :)
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

NeroBurner, 2022-09-04:
Ludovic (FederAndInk) recommended the cmake module https://github.com/lethal-guitar/RigelEngine/blob/master/cmake/Modules/FindFilesystem.cmake to handle the linking of `stdc++fs` if needed. This should make `littlefs-do` compile and link successfully again on g++-8

NeroBurner, 2022-09-04:
@RazRosman could you please try https://github.com/InfiniTimeOrg/InfiniSim/pull/58 and see if that works for you?

RazRosman, 2022-09-09:
Yes, I can check #58, but what do I do with that file

RazRosman, 2022-09-09:
I tried another thing.
I installed gcc-9.1 and now I'm getting another error when creating the build directory

pi@raspberrypi:~/InfiniSim $ export CC=/opt/gcc-9.1.0/bin
pi@raspberrypi:~/InfiniSim $ export CXX=/opt/gcc-9.1.0/bin
pi@raspberrypi:~/InfiniSim $ cmake -S . -B build
-- Using CMake version 3.16.3
-- InfiniTime PROJECT_VERSION extracted: 1.10.0
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Check for working C compiler: /opt/gcc-9.1.0/bin
-- Check for working C compiler: /opt/gcc-9.1.0/bin -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/opt/gcc-9.1.0/bin"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/pi/InfiniSim/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_c7f58/fast && /usr/bin/make -f CMakeFiles/cmTC_c7f58.dir/build.make CMakeFiles/cmTC_c7f58.dir/build
    make[1]: Entering directory '/home/pi/InfiniSim/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_c7f58.dir/testCCompiler.c.o
    /opt/gcc-9.1.0/bin    -o CMakeFiles/cmTC_c7f58.dir/testCCompiler.c.o   -c /home/pi/InfiniSim/build/CMakeFiles/CMakeTmp/testCCompiler.c
    make[1]: execvp: /opt/gcc-9.1.0/bin: Permission denied
    make[1]: *** [CMakeFiles/cmTC_c7f58.dir/build.make:66: CMakeFiles/cmTC_c7f58.dir/testCCompiler.c.o] Error 127
    make[1]: Leaving directory '/home/pi/InfiniSim/build/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_c7f58/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:18 (project)

-- Configuring incomplete, errors occurred!
See also "/home/pi/InfiniSim/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/InfiniSim/build/CMakeFiles/CMakeError.log".

Should I resolve this issue with gcc-9.1 or get back to 8 and solve the previous issue?

Thanks,
Raz

NeroBurner, 2022-09-11:
you need to pass a full path to the `gcc` and `g++` compiler executables. Probably the following

```sh
export CC=/opt/gcc-9.1.0/bin/gcc
export CXX=/opt/gcc-9.1.0/bin/g++
```

If you can resolve it with gcc-9.1 I can close the PR and just say we need a compiler supporting `std::filesystem`

RazRosman, 2022-09-15:
Using gcc 9.1 and setting the executables work, but now I'm getting the following error
pi@raspberrypi:~/InfiniSim $ cmake -S . -B build
-- Using CMake version 3.16.3
-- InfiniTime PROJECT_VERSION extracted: 1.10.0
-- The C compiler identification is GNU 9.1.0
-- The CXX compiler identification is GNU 9.1.0
-- Check for working C compiler: /opt/gcc-9.1.0/bin/gcc-9.1
-- Check for working C compiler: /opt/gcc-9.1.0/bin/gcc-9.1 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/gcc-9.1.0/bin/g++-9.1
-- Check for working CXX compiler: /opt/gcc-9.1.0/bin/g++-9.1 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using default C++ standard 17
-- Using MONITOR_ZOOM=1
-- add subdirectory /home/pi/InfiniSim/InfiniTime/src/displayapp/fonts for 'infinitime_fonts' target
-- Using /usr/local/bin/lv_font_conv to generate font files
-- Found Python3: /usr/bin/python3.7 (found version "3.7.3") found components: Interpreter 
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find SDL2 (missing: SDL2_LIBRARY)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindSDL2.cmake:79 (find_package_handle_standard_args)
  CMakeLists.txt:287 (find_package)

RazRosman, 2022-09-15:
So I downloaded SDL2 and I'm getting an error message when building

 CC     build/SDL_hidapi.lo
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c: In function ‘HIDAPI_UpdateDiscovery’:
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c:435:73: error: ‘NAME_MAX’ undeclared (first use in this function)
  435 |                 char enough_for_inotify[sizeof (struct inotify_event) + NAME_MAX + 1];
      |                                                                         ^~~~~~~~
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c:435:73: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:365: build/SDL_hidapi.lo] Error 1
/bin/bash ./build-scripts//updaterev.sh
  CC     build/SDL_hidapi.lo
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c: In function ‘HIDAPI_UpdateDiscovery’:
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c:435:73: error: ‘NAME_MAX’ undeclared (first use in this function)
  435 |                 char enough_for_inotify[sizeof (struct inotify_event) + NAME_MAX + 1];
      |                                                                         ^~~~~~~~
/home/pi/Downloads/SDL2-2.24.0/src/hidapi/SDL_hidapi.c:435:73: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:365: build/SDL_hidapi.lo] Error 1

Used the following link
https://forums.raspberrypi.com/viewtopic.php?t=58180

NeroBurner, 2022-09-16:
Installing the libsdl development Debian package as written in the readme didn't work?
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

RazRosman, 2022-09-16:
no.

pi@raspberrypi:~ $ sudo apt install -y cmake libsdl2-dev g++ npm
Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++ is already the newest version (4:8.3.0-1+rpi2).
npm is already the newest version (5.8.0+ds6-4+deb10u2).
cmake is already the newest version (3.16.3-3~bpo10+1).
libsdl2-dev is already the newest version (2.0.9+dfsg1-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

it's already installed.

RazRosman, 2022-09-17:
Managed to solve it and also to install the ZLIB library.
Now finally I'm running cmake -S . -B build
it's almost finished and now I'm getting the following error:

-- Using CMake version 3.16.3
-- InfiniTime PROJECT_VERSION extracted: 1.10.0
-- Using default C++ standard 17
-- Using MONITOR_ZOOM=1
-- add subdirectory /home/pi/InfiniSim/InfiniTime/src/displayapp/fonts for 'infinitime_fonts' target
-- Using /usr/local/bin/lv_font_conv to generate font files
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Symbol prefix: 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
M_LIBRARY
    linked by target "png-fix-itxt" in directory /home/pi/InfiniSim/libpng
    linked by target "png" in directory /home/pi/InfiniSim/libpng
    linked by target "png_static" in directory /home/pi/InfiniSim/libpng

-- Configuring incomplete, errors occurred!
See also "/home/pi/InfiniSim/build/CMakeFiles/CMakeOutput.log".

NeroBurner, 2022-09-17:
I updated the `libpng` module https://github.com/InfiniTimeOrg/InfiniSim/pull/65 to get the following fix in, which might be the culprit for you https://github.com/glennrp/libpng/commit/763c77e628e2c78b4c422bb7e26327d28ee54ad4

please update InfiniSim (and the libpng submodule if it doesn't itself)

RazRosman, 2022-09-20:
What do you mean by update Infinisim and libpng?
I copied the content of the file in the fix and got the following errors:

-- Symbol prefix: 
CMake Error: File /home/pi/InfiniSim/scripts/checksym.awk does not exist.
CMake Error at CMakeLists.txt:300 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/scripts/options.awk does not exist.
CMake Error at CMakeLists.txt:304 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/scripts/dfn.awk does not exist.
CMake Error at CMakeLists.txt:308 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:695 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:702 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:705 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:708 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:711 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:714 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:717 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:720 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:723 (png_add_test)

[... 132 lines cut ...]
CMake Error: File /home/pi/InfiniSim/scripts/test.cmake.in does not exist.
CMake Error at CMakeLists.txt:672 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  CMakeLists.txt:839 (png_add_test)

CMake Error: File /home/pi/InfiniSim/scripts/genchk.cmake.in does not exist.
CMake Error at CMakeLists.txt:910 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/scripts/genout.cmake.in does not exist.
CMake Error at CMakeLists.txt:913 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/scripts/gensrc.cmake.in does not exist.
CMake Error at CMakeLists.txt:916 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/libpng.pc.in does not exist.
CMake Error at CMakeLists.txt:935 (configure_file):
  configure_file Problem configuring file

CMake Error: File /home/pi/InfiniSim/libpng-config.in does not exist.
CMake Error at CMakeLists.txt:939 (configure_file):
  configure_file Problem configuring file

RazRosman, 2022-09-20:
Ok.
First build - WORKING :)))))))))))))))))))
I cloned it again

RazRosman, 2022-09-20:
Second build crashed.

Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_line.c.o
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/usr/local/lib/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/usr/local/lib/node_modules/lv_font_conv/lib/convert.js:25:35)
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_mask.c.o
Traceback (most recent call last):
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/lv_font_conv', '--size', '42', '--output', 'jetbrains_mono_42.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/JetBrainsMono-Regular.ttf', '--range', '0x25, 0x2b, 0x2d, 0x30-0x3a']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_42.dir/build.make:61: fonts/jetbrains_mono_42.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:540: fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_42.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_rect.c.o
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_triangle.c.o
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_buf.c.o
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/usr/local/lib/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/usr/local/lib/node_modules/lv_font_conv/lib/convert.js:25:35)
Traceback (most recent call last):
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/lv_font_conv', '--size', '80', '--output', 'jetbrains_mono_extrabold_compressed.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/JetBrainsMono-ExtraBold.ttf', '--range', '0x30-0x3a']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/build.make:61: fonts/jetbrains_mono_extrabold_compressed.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:345: fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_extrabold_compressed.dir/all] Error 2
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_cache.c.o
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_decoder.c.o
[ 11%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font.c.o
[ 11%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_dejavu_16_persian_hebrew.c.o
[ 11%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_fmt_txt.c.o
[ 11%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_loader.c.o
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_10.c.o
TypeError: this.src.glyphs.map(...).flat is not a function
    at new Font (/usr/local/lib/node_modules/lv_font_conv/lib/font/font.js:38:99)
    at new LvFont (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/lv_font.js:16:5)
    at Object.write_images [as lvgl] (/usr/local/lib/node_modules/lv_font_conv/lib/writers/lvgl/index.js:12:16)
    at convert (/usr/local/lib/node_modules/lv_font_conv/lib/convert.js:25:35)
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_12.c.o
Traceback (most recent call last):
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 75, in <module>
    main()
  File "/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/generate.py", line 67, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/bin/lv_font_conv', '--size', '76', '--output', 'jetbrains_mono_76.c', '--bpp', '1', '--format', 'lvgl', '--no-compress', '--font', '/home/pi/InfiniSim/InfiniTime/src/displayapp/fonts/JetBrainsMono-Light.ttf', '--range', '0x25, 0x2D, 0x2F, 0x30-0x3a']' returned non-zero exit status 1.
make[2]: *** [fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_76.dir/build.make:61: fonts/jetbrains_mono_76.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:372: fonts/CMakeFiles/infinitime_fonts_jetbrains_mono_76.dir/all] Error 2
...
...
...
[ 46%] Building CXX object CMakeFiles/sim-base.dir/sim/libraries/timer/app_timer.cpp.o
[ 47%] Linking CXX static library libsim-base.a
[ 47%] Built target sim-base
make: *** [Makefile:130: all] Error 2

RazRosman, 2022-09-20:
It was a node.js issue.
I was using v10 and installed v14.
second build finished successfully with warnings.
[100%] Linking CXX executable infinisim
[100%] Built target infinisim

RazRosman, 2022-09-20:
Infinisim Working :))))))))))))))))))
THank you very much for the help

NeroBurner, 2022-09-21:
happy it works :tada:

closed 2022-09-21 by NeroBurner

devnoname120, 2023-01-18:
Crosspost from https://github.com/InfiniTimeOrg/InfiniSim/pull/58

--------

I had the following error when trying to compile `InfiniSim`:

```shell
parallels@debian-gnu-linux-10:~/InfiniSim$ cmake --build build -j 4
-- Using CMake version 3.13.4
-- InfiniTime PROJECT_VERSION extracted: 1.11.0
-- Using default C++ standard 17
-- Using MONITOR_ZOOM=1
-- add subdirectory /home/parallels/InfiniSim/InfiniTime/src/displayapp/fonts for 'infinitime_fonts' target
-- Using /home/parallels/InfiniSim/node_modules/.bin/lv_font_conv to generate font files
-- Symbol prefix: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/InfiniSim/build
[  0%] Built target scripts_symbols_out
[  0%] Built target pngprefix_h
[  1%] Built target pnglibconf_c
[  2%] Built target scripts_pnglibconf_c
[  3%] Built target littlefs
[  3%] Built target infinitime_fonts_open_sans_light
[  4%] Built target QCBOR
[  4%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  5%] Built target infinitime_fonts_lv_font_sys_48
[  5%] Built target infinitime_fonts_jetbrains_mono_42
[  5%] Built target infinitime_fonts_lv_font_navi_80
[ 47%] Built target sim-base
[ 48%] Built target infinitime_fonts_jetbrains_mono_76
[ 48%] Built target infinitime_fonts_jetbrains_mono_bold_20
[ 50%] Built target scripts_symbols_chk
[ 50%] Built target miniz
[ 51%] Built target pnglibconf_out
[ 55%] Built target infinitime_fonts
[ 55%] Built target scripts_prefix_out
[ 56%] Built target pnglibconf_h
[ 56%] Linking CXX executable littlefs-do
[ 57%] Built target scripts_sym_out
[ 58%] Built target scripts_intprefix_out
[ 60%] Built target scripts_vers_out
[ 61%] Built target gensym
[ 61%] Built target genvers
[ 64%] Built target genfiles
[ 71%] Built target png_static
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `command_cp(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool)':
littlefs-do-main.cpp:(.text+0x24d8): undefined reference to `std::filesystem::file_size(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `mkdir_path(std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text+0x38b4): undefined reference to `std::filesystem::__cxx11::path::parent_path() const'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `command_res(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, bool)':
littlefs-do-main.cpp:(.text+0x3d64): undefined reference to `std::filesystem::file_size(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: littlefs-do-main.cpp:(.text+0x4130): undefined reference to `std::filesystem::__cxx11::path::parent_path() const'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::path(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, std::filesystem::__cxx11::path::format)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathC2EONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS1_6formatE[_ZNSt10filesystem7__cxx114pathC5EONSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS1_6formatE]+0x4c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::operator/=(std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x38): undefined reference to `std::filesystem::__cxx11::path::has_filename() const'
/usr/bin/ld: littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx114pathdVERKS1_[_ZNSt10filesystem7__cxx114pathdVERKS1_]+0x90): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::is_absolute() const':
littlefs-do-main.cpp:(.text._ZNKSt10filesystem7__cxx114path11is_absoluteEv[_ZNKSt10filesystem7__cxx114path11is_absoluteEv]+0x10): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::operator==(std::filesystem::__cxx11::path const&, std::filesystem::__cxx11::path const&)':
littlefs-do-main.cpp:(.text._ZNSt10filesystem7__cxx11eqERKNS0_4pathES3_[_ZNSt10filesystem7__cxx11eqERKNS0_4pathES3_]+0x18): undefined reference to `std::filesystem::__cxx11::path::compare(std::filesystem::__cxx11::path const&) const'
/usr/bin/ld: CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o: in function `std::filesystem::__cxx11::path::extension() const':
littlefs-do-main.cpp:(.text._ZNKSt10filesystem7__cxx114path9extensionEv[_ZNKSt10filesystem7__cxx114path9extensionEv]+0x18): undefined reference to `std::filesystem::__cxx11::path::_M_find_extension() const'
[... cut ...]
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Symbol prefix: 
-- Looking for C++ include filesystem
-- Looking for C++ include filesystem - found
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED - Failed
-- Performing Test CXX_FILESYSTEM_STDCPPFS_NEEDED
-- Performing Test CXX_FILESYSTEM_STDCPPFS_NEEDED - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parallels/InfiniSim/build
```