InfiniTimeOrg/InfiniSim #72: Segmentation fault when using the simulator

rcaneill, 2022-10-17:
Hello,
thanks for the simulator! I however have a problem: the compilation is a success, but when I run the simulator for almost every thing I try I get a `Segmentation fault (core dumped)`. E.g. if I try to slide up to get the list of applications, or slide down to get the notifications. If I slide right, I access the configuration menu, where I can change the brightness or the notification bell, but if I click on the gear or the light I get the segmentation fault again.

I tried v1.10.0, v1.11.0 and latest, with InfiniTime v1.10.0 or latest, but without any change.

I run a Linux Mint laptop, cmake version is 3.22.1, g++ version 11.2.0, python3.10.6. I don't know what else I could give as logs.
Thanks!

NeroBurner, 2022-10-17:
can you pastebin your (clean) cmake configure and build logs? maybe there is something in there

rcaneill, 2022-10-17:
```
-- Using CMake version 3.22.1
-- InfiniTime PROJECT_VERSION extracted: 1.10.0
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using default C++ standard 17
-- Using MONITOR_ZOOM=3
-- add subdirectory /home/romain/Downloads/InfiniTime-1.10.0/src/displayapp/fonts for 'infinitime_fonts' target
-- Using /home/romain/PineTime/InfiniSim/node_modules/.bin/lv_font_conv to generate font files
-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Symbol prefix: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/romain/PineTime/InfiniSim/build

[  0%] Generating pnglibconf.c
[  1%] Generating scripts/pnglibconf.c
[  1%] Generating pngprefix.h
[  1%] Generating scripts/symbols.out
[  1%] Built target pngprefix_h
[  1%] Built target pnglibconf_c
[  1%] Building C object CMakeFiles/QCBOR.dir/home/romain/Downloads/InfiniTime-1.10.0/src/libs/QCBOR/src/ieee754.c.o
[  1%] Built target scripts_pnglibconf_c
[  1%] Built target scripts_symbols_out
[  2%] Generating open_sans_light.c
[  2%] Building C object CMakeFiles/QCBOR.dir/home/romain/Downloads/InfiniTime-1.10.0/src/libs/QCBOR/src/qcbor_decode.c.o
[  2%] Generating jetbrains_mono_42.c
[  2%] Generating jetbrains_mono_76.c
[  2%] Building C object CMakeFiles/QCBOR.dir/home/romain/Downloads/InfiniTime-1.10.0/src/libs/QCBOR/src/qcbor_encode.c.o
[  3%] Building C object CMakeFiles/QCBOR.dir/home/romain/Downloads/InfiniTime-1.10.0/src/libs/QCBOR/src/qcbor_err_to_str.c.o
[  3%] Building C object CMakeFiles/QCBOR.dir/home/romain/Downloads/InfiniTime-1.10.0/src/libs/QCBOR/src/UsefulBuf.c.o
[  3%] Linking C static library libQCBOR.a
[  3%] Built target QCBOR
[  4%] Generating jetbrains_mono_bold_20.c
[  4%] Built target infinitime_fonts_open_sans_light
[  4%] Built target infinitime_fonts_jetbrains_mono_42
[  4%] Generating jetbrains_mono_extrabold_compressed.c
[  4%] Generating lv_font_navi_80.c
[  4%] Built target infinitime_fonts_jetbrains_mono_76
[  4%] Generating lv_font_sys_48.c
patching file jetbrains_mono_bold_20.c
patching file jetbrains_mono_bold_20.c
[  4%] Built target infinitime_fonts_jetbrains_mono_bold_20
[  4%] Generating pnglibconf.out
[  4%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  5%] Generating scripts/symbols.chk
[  5%] Built target pnglibconf_out
[  5%] Built target infinitime_fonts_lv_font_sys_48
[  5%] Generating pnglibconf.h
[  5%] Built target scripts_symbols_chk
[  5%] Generating scripts/prefix.out
[  6%] Built target pnglibconf_h
[  7%] Generating scripts/sym.out
[  7%] Generating scripts/intprefix.out
[  7%] Built target scripts_prefix_out
[... 302 lines cut ...]
[ 97%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/components/motor/MotorController.cpp.o
In file included from /home/romain/Downloads/InfiniTime-1.10.0/src/components/datetime/DateTimeController.cpp:3:
/home/romain/Downloads/InfiniTime-1.10.0/src/components/datetime/DateTimeController.cpp: In member function ‘void Pinetime::Controllers::DateTime::SetTime(uint16_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint32_t)’:
/home/romain/Downloads/InfiniTime-1.10.0/src/components/datetime/DateTimeController.cpp:47:16: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘Pinetime::Controllers::DateTime::Months’ [-Wformat=]
   47 |   NRF_LOG_INFO("* %d %d %d ", this->day, this->month, this->year);
      |                ^~~~~~~~~~~~~             ~~~~~~~~~~~
      |                                                |
      |                                                Pinetime::Controllers::DateTime::Months
/home/romain/PineTime/InfiniSim/sim/libraries/log/nrf_log.h:56:66: note: in definition of macro ‘NRF_LOG_INFO’
   56 | #define NRF_LOG_INFO(...)         do {printf("info:  "); printf( __VA_ARGS__ ); printf("\n"); } while(false);
      |                                                                  ^~~~~~~~~~~
/home/romain/Downloads/InfiniTime-1.10.0/src/components/datetime/DateTimeController.cpp:47:23: note: format string is defined here
   47 |   NRF_LOG_INFO("* %d %d %d ", this->day, this->month, this->year);
      |                      ~^
      |                       |
      |                       int
[ 98%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/components/timer/TimerController.cpp.o
[ 98%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/drivers/Spi.cpp.o
[ 98%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/drivers/St7789.cpp.o
[100%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/touchhandler/TouchHandler.cpp.o
[100%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/systemtask/SystemTask.cpp.o
[100%] Building CXX object CMakeFiles/infinisim.dir/home/romain/Downloads/InfiniTime-1.10.0/src/systemtask/SystemMonitor.cpp.o
[100%] Linking CXX executable infinisim
[100%] Built target infinisim
```

NeroBurner, 2022-10-17:
Are you on the latest version of `InfiniSim`? I don't see the `littlefs-do` command being compiled. And you'll need the `lv_img_conv` tool installed to create the new ressource.zip file (see README.md for installation instructions)

rcaneill, 2022-10-17:
I don't know why, but running with the status windows raised directly a seg fault, I then tried without the status window and it worked fine. I then retried with the status window and it works fine...

NeroBurner, 2022-10-17:
That's odd ... Closing this issue for now. Please reopen if the segfaults return

closed 2022-10-17 by NeroBurner

SteveAmor, 2023-10-05:
I have the same problem.  ```--hide-status``` worked for me.  Strangely, I then ran the simulator without the ```hide-status``` straight after and it didn't get the segfault.  Closing the terminal window sometimes fixed it.  Looks like there is a race condition?  I'm on an old T420 running Ubuntu 22.04 with https://github.com/InfiniTimeOrg/InfiniSim/commit/02f2cc5a05292f16ced26d2577c5479e9528ba78 InfiniSim.