InfiniTimeOrg/InfiniSim #92: Add warning that std::chrono::system_clock is not supported on hardware even though it may work fine in the sim

kestara, 2023-03-01:
(This confusion has happened at least two times [link 1](https://github.com/InfiniTimeOrg/InfiniTime/discussions/1582) [link 2](https://github.com/InfiniTimeOrg/InfiniTime/discussions/1656))

Idea: If use of `std::chrono::system_clock` is detected (unsure how) then add a warning in the console that on the watch this will result in undefined behaviour.

NeroBurner, 2023-03-16:
I very much like the idea, but sadly I don't know how to pull it off.

lmamane, 2024-10-27:
Here's a proof of concept. Works on GNU/Linux at least, other platforms
 to be tested.

[stdchrono_abort.diff.txt](https://github.com/user-attachments/files/17532874/stdchrono_abort.diff.txt)

NeroBurner, 2024-10-27:
That looks interesting to say the least! Could you open a PR with those changes for easier workflow on my side and attribution for your work? 🙇‍♂️

lmamane, 2024-11-09:
Assuming my pull request #2162 in InfiniTime is merged, this will become moot and can be closed.

NeroBurner, 2025-06-10:
Direct link to mentioned PR: https://github.com/InfiniTimeOrg/InfiniTime/pull/2162