InfiniTimeOrg/InfiniSim #99: Cannot build simulator

bschau, 2023-04-14:
I am following the guide.  However, in the "cmake --build build" steps it fails with:

[ 68%] Linking C static library libinfinitime_fonts.a
[ 68%] Built target infinitime_fonts
[ 68%] Generate fonts and images for resource package
/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1690
  const sourceMap = JSON.parse(sourceMapText);
                         ^
SyntaxError: "undefined" is not valid JSON
    at JSON.parse (<anonymous>)
    at updateSourceMap (/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1690:26)
    at updateOutput (/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1640:5)
    at Object.compile (/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1460:20)
    at Module.m._compile (/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1617:30)
    at Module._extensions..js (node:internal/modules/cjs/loader:1329:10)
    at Object.require.extensions.<computed> [as .ts] (/home/bs/.npm-global/lib/node_modules/ts-node/src/index.ts:1621:12)
    at Module.load (node:internal/modules/cjs/loader:1133:32)
    at Function.Module._load (node:internal/modules/cjs/loader:972:12)
    at Module.require (node:internal/modules/cjs/loader:1157:19)
Traceback (most recent call last):
  File "/home/bs/InfiniTime/InfiniSim/InfiniTime/src/resources/generate-img.py", line 56, in <module>
    main()
  File "/home/bs/InfiniTime/InfiniSim/InfiniTime/src/resources/generate-img.py", line 51, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/bs/.npm-global/bin/lv_img_conv', '/home/bs/InfiniTime/InfiniSim/InfiniTime/src/resources/images/pine_logo.png', '--force', '--output-file', 'pine_small.bin', '--color-format', 'CF_TRUE_COLOR_ALPHA', '--output-format', 'bin', '--binary-format', 'ARGB8565_RBSWAP']' returned non-zero exit status 1.
gmake[2]: *** [resources/CMakeFiles/GenerateResources.dir/build.make:73: resources/CMakeFiles/GenerateResources] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:568: resources/CMakeFiles/GenerateResources.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2

I am using Node.js v19 and on the lastest Pop!_OS (based on ubuntu 22.04 lts).

What can I do to debug this?

Itai-Nelken, 2023-04-14:
I had the same problem (on pop os as well), the fix is to install the exact versions of `lv_font_conv` and `lv_img_conv` that are in the readme:
```bash
npm install lv_font_conv@1.5.2
npm install ts-node@10.9.1 @swc/core lv_img_conv@0.3.0
```
Make sure to remove the old versions first by deleting the node_modules folder in the InfiniSim folder.

bschau, 2023-04-14:
Hmm - I tried your suggestions with no luck. Still the same error.
I even removed the InfiniSim folder entirely and started all over :-(

In essence it tries to run:

home/bs/InfiniTime/InfiniSim/node_modules/.bin/lv_img_conv /home/bs/InfiniTime/InfiniSim/InfiniTime/src/resources/images/pine_logo.png --force --output-file pine_small.bin --color-format CF_TRUE_COLOR_ALPHA --output-format bin --binary-format ARGB8565_RBSWAP
/home/bs/InfiniTime/InfiniSim/node_modules/ts-node/src/index.ts:1690
  const sourceMap = JSON.parse(sourceMapText);

The file complaining seems to be dynamically generated as ts-node/src/index.ts does not exist.  So whatever is wrong will be hard to find out :-(

Thanks for helping me.

bschau, 2023-04-14:
Ok, so, now I have some progress.  I ran:

cmake -S . -B build -DBUILD_RESOURCES=OFF

(to skip resource building) and then

cmake --build build

.. and it built the simulator.
However, starting the simulator:

bs@dharma ~/InfiniTime/InfiniSim (*** MAIN ***)
$ ./build/infinisim 
/home/bs/InfiniTime/InfiniSim/InfiniTime/src/libs/littlefs/lfs.c:1070:error: Corrupted dir pair at {0x0, 0x1}
Segmentation fault (core dumped)

.. if I start with the --hide-status flag it will launch!  Yay!

Then, if I run ./build/infinisim (without the --hide-status flag) it will in  ~ 2 out of 3 launches break with an segmentation fault:

$ ./build/infinisim
Segmentation fault (core dumped)

Launching with the --hide-status flags seems to always work.

My question now is: what am I missing when the resources file have not been built?
My ultimate goal is to convert one of my old Pebble watchfaces to run on the PineTime ...

NeroBurner, 2023-05-03:
is this still an issue?

bschau, 2023-05-03:
I haven't tried & I moved on.

closed 2023-05-03 by bschau