InfiniTimeOrg/InfiniSim #109: typescript: "startIndex ?? 0" is not recognized

araud, 2023-07-07:
[build] /mnt/c/Work/astute.watch/InfiniSim/node_modules/typescript/lib/typescript.js:139
[build]     for (let i = startIndex ?? 0; i < array.length; i++) {
[build]                              ^
[build] 
[build] SyntaxError: Unexpected token '?'
[build]     at wrapSafe (internal/modules/cjs/loader.js:915:16)
[build]     at Module._compile (internal/modules/cjs/loader.js:963:27)
[build]     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
[build]     at Module.load (internal/modules/cjs/loader.js:863:32)
[build]     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
[build]     at Module.require (internal/modules/cjs/loader.js:887:19)
[build]     at require (internal/modules/cjs/helpers.js:74:18)
[build]     at attemptRequireWithV8CompileCache (/mnt/c/Work/astute.watch/InfiniSim/node_modules/ts-node/dist/util.js:104:16)
[build]     at loadCompiler (/mnt/c/Work/astute.watch/InfiniSim/node_modules/ts-node/dist/configuration.js:237:56)
[build]     at resolveAndLoadCompiler (/mnt/c/Work/astute.watch/InfiniSim/node_modules/ts-node/dist/configuration.js:226:16)
[build] Traceback (most recent call last):
[build]   File "/mnt/c/Work/astute.watch/InfiniSim/InfiniTime/src/resources/generate-img.py", line 56, in <module>
[build]     main()
[build]   File "/mnt/c/Work/astute.watch/InfiniSim/InfiniTime/src/resources/generate-img.py", line 51, in main
[build]     subprocess.check_call(line)
[build]   File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
[build]     raise CalledProcessError(retcode, cmd)
[build] subprocess.CalledProcessError: Command '['/mnt/c/Work/astute.watch/InfiniSim/node_modules/.bin/lv_img_conv', '/mnt/c/Work/astute.watch/InfiniSim/InfiniTime/src/resources/images/pine_logo.png', '--force', '--output-file', 'pine_small.bin', '--color-format', 'CF_TRUE_COLOR_ALPHA', '--output-format', 'bin', '--binary-format', 'ARGB8565_RBSWAP']' returned non-zero exit status 1.
[build] gmake[2]: *** [resources/CMakeFiles/GenerateResources.dir/build.make:73: resources/CMakeFiles/GenerateResources] Error 1
[build] gmake[1]: *** [CMakeFiles/Makefile2:615: resources/CMakeFiles/GenerateResources.dir/all] Error 2
[build] gmake: *** [Makefile:136: all] Error 2
[image]
InfiniSim$ node -v
v12.22.9
InfiniSim$ npm -v
8.5.1

NeroBurner, 2023-07-07:
Please provide additional info about the commands to reproduce this issue. How did you setup your build environment. What os is used and so on.

With the current information I have no idea how to start debugging the error

araud, 2023-07-08:
Hi @NeroBurner ! Thanks for the fast reply! I use WSL2 (Ubuntu) and just follow the steps:
[image]
Then config & build 100% same as told there.

Someone told me that building resources also failed for him.
Disabling BUILD_RESOURCES helps to build well indeed.

hrmckay, 2023-07-16:
I had exactly the same error.  The problem seems to be the level of Javascript.  On my system the node -v command gave v12.22.9 which is old.  I installed NVM and used it to install the latest version of node.js, v20.4.0.

I reran the cmake command and got a different error, but it suggested running npm rebuild.  I did that and the next run of cmake worked perfectly and infinisim was built.

Hugh

NeroBurner, 2023-07-16:
Thanks for the reply. It seems the lv-font-fonf script updated and now needs at least node v14+

[image]

I think our readme should be updated to reflect that change

closed 2023-09-17 by NeroBurner