InfiniTimeOrg/InfiniSim #110: Display is skewed during and after vertical transitions

mantasg, 2023-07-08:
I've built and tried using InfiniSim on Ubuntu 20.04 following standard instructions in README file. It builds and starts fine. However, whenever I perform vertical swipe gestures (either using mouse or arrow keys) the display gets skewed. Horizontal gestures seem to work fine.

I'm trying to debug it myself, but does anyone have any thoughts of what could be causing this?

I'm including GIF recording of the issue:
- It starts with the screen off
- I activate the screen with right mouse click
- I swipe right and left to demonstrate working horizontal transitions
- Then I swipe up and down resulting in skewed display
- I then turn off screen with right click and turn it back on with the display going back to normal

[image]

I'm happy to provide any additional information.

mantasg, 2023-07-08:
Adding another preview

https://github.com/InfiniTimeOrg/InfiniSim/assets/1678730/9a6f90d9-bbcd-4030-af7a-ea89127d7dac

mantasg, 2023-07-08:
Adding another preview

[image]

mantasg, 2023-07-12:
I've managed to get it to a workable state using this change. It doesn't show vertical scroll animations, but at least the screens are not distorted. Still would be interesting to know what's going on there.

```
--- a/sim/displayapp/LittleVgl.cpp
+++ b/sim/displayapp/LittleVgl.cpp
@@ -141,8 +141,10 @@ void LittleVgl::InitFileSystem() {
 void LittleVgl::SetFullRefresh(FullRefreshDirections direction) {
   if (scrollDirection == FullRefreshDirections::None) {
     scrollDirection = direction;
-    if (scrollDirection == FullRefreshDirections::Down) {
+    if (scrollDirection == FullRefreshDirections::Up) {
       lv_disp_set_direction(lv_disp_get_default(), 1);
+    } else if (scrollDirection == FullRefreshDirections::Down) {
+      lv_disp_set_direction(lv_disp_get_default(), 0);
     } else if (scrollDirection == FullRefreshDirections::Right) {
       lv_disp_set_direction(lv_disp_get_default(), 2);
     } else if (scrollDirection == FullRefreshDirections::Left) {
```

NeroBurner, 2023-08-18:
Can you provide more info about your system hardware. I'm especially interested in CPU, GPU and used drivers for the GPU