InfiniTimeOrg/InfiniSim #112: Support launching an app/screen directly

jackwilsdon, 2023-07-30:
As the title says - it'd be good if an app/screen could be launched directly at startup. This would help improve the feedback loop whilst developing new features.

NeroBurner, 2023-08-18:
This should be already possible as the number key already are bound to specific screens. Would Elit be enough for your usecase to programatically locally "rebind" one of the number keys?

NeroBurner, 2023-09-11:
Closed due to inactivity

closed 2023-09-11 by NeroBurner

jackwilsdon, 2023-09-11:
Apologies for the late reply on this. I didn't realise specific apps could be launched via the number keys - that's definitely useful. Is it possible to launch any apps after app 9? If not, it seems like a command line flag might still be useful (and would be useful for development anyway, allowing for launching straight into an app from the command line).

jackwilsdon, 2023-09-11:
> A general switch wouldn't be feasible as we have more than 20 screens, and which ones do we select?

Unless I'm misunderstanding, what prevents a command line flag like `-screen 9` being added? Having to refer to the screens by ID is not the most ideal, but it keeps it flexible (and means that a list of screen names isn't heeded).

NeroBurner, 2023-09-11:
I'm sorry, I may am slow at the uptake here. Which screen would be launched by defining `-screen 9`? Could you describe what actions you would take in your scenario and what you would then expect to happen?