InfiniTimeOrg/InfiniSim #116: Building issue on Debian wsl VM running on top of windows 11

coffandro, 2023-08-17:
I'm trying to install this but when i run `cmake -S . -B build` it gives me the following error 

`CMake Error at InfiniTime/src/resources/CMakeLists.txt:6 (find_program):
  Could not find LV_IMG_CONV using the following names: lv_img_conv`
after
`-- Using /usr/local/bin/lv_font_conv to generate font files`

Here's the Output log
[CMakeOutput.log](https://github.com/InfiniTimeOrg/InfiniSim/files/12369073/CMakeOutput.log)

NeroBurner, 2023-08-18:
You need to install build dependencies as documented in the Readme

https://github.com/InfiniTimeOrg/InfiniSim#build-dependencies

I think the minimum node version is 14 now. Should fix that 🙈

closed 2023-08-18 by NeroBurner

coffandro, 2023-08-18:
I did that.
I ran the following command listed in order:

1. git clone --recursive https://github.com/InfiniTimeOrg/InfiniSim.git
2. cd InfiniSim
3. sudo apt update
4. git submodule update --init --recursive (better safe than sorry)
5. sudo apt install -y cmake libsdl2-dev g++ npm
6. npm install lv_font_conv@1.5.2
7. npm install ts-node@10.9.1 @swc/core lv_img_conv@0.3.0
8. cmake -S . -B build
Then it gives me my errors. I did install all the build dependencies and checked my node.js version which was v18

coffandro, 2023-08-18:
running cmake -S . -B build -DBUILD_RESOURCES=OFF somehow fixed it???

NeroBurner, 2023-08-18:
Then I'd check the lv-ing-conv dependencies. From their readme try installing the required system packages

```sh
sudo apt install gcc make g++ pkg-config libpixman-1-dev libcairo2-dev libpango1.0-dev libjpeg8-dev libgif-dev
```

coffandro, 2023-08-18:
libjpeg8-dev didn't want to install but i found a file with it and got it installed and working. then i installed lv_img_conv with their install instructions. i've updated everything and done just about anything i could think of and it still isn't working

NeroBurner, 2023-08-18:
Try to confirm that running the lv_img_conv command on a PNG manually succeeds

coffandro, 2023-09-17:
I fixed it, i was using the wrong version of NodeJS like you said lol

NeroBurner, 2023-09-17:
Happy you got it fixed!

coffandro, 2023-09-17:
[at least this issue though i it then broke on the next step lol](https://github.com/InfiniTimeOrg/InfiniSim/issues/121)