InfiniTimeOrg/InfiniSim #120: Build failing

getlogname, 2023-09-10:
Hi, I'm trying to build the simulator on Fedora but without success. I also tried on a fresh Ubuntu VM but had similar result, I'm obviously doing something wrong, but what?

Steps:
1) `git clone --recursive https://github.com/InfiniTimeOrg/InfiniSim.git`
2) `cd InfiniSim/`
3) `git submodule update --init --recursive`
4) `sudo dnf install cmake SDL2-devel gcc zlib-devel npm`
5) `npm install lv_font_conv@1.5.2`
6) `npm install ts-node@10.9.1 @swc/core lv_img_conv@0.3.0`
7) `cmake -S . -B build` Here I get a depreciation warning but I'm already using cmake 3.27.4 which is the latest version.
8) `cmake --build build -j4`

<details>
  <summary>Depreciation warning</summary>

```
-- Using CMake version 3.27.4
-- InfiniTime PROJECT_VERSION extracted: 1.13.0
-- The C compiler identification is GNU 13.2.1
-- The CXX compiler identification is GNU 13.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using default C++ standard 17
-- Using MONITOR_ZOOM=1
-- add subdirectory /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/InfiniTime/src/displayapp/fonts for 'infinitime_fonts' target
-- Using /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/node_modules/.bin/lv_font_conv to generate font files
-- Found Python3: /usr/bin/python3.11 (found version "3.11.4") found components: Interpreter 
-- add subdirectory /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/InfiniTime/src/resources for 'GenerateResources' target
-- Using /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/node_modules/.bin/lv_font_conv to generate font files
-- Using /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/node_modules/.bin/lv_img_conv to generate font files
-- Looking for C++ include filesystem
-- Looking for C++ include filesystem - found
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED - Success
CMake Deprecation Warning at libpng/CMakeLists.txt:27 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at libpng/CMakeLists.txt:28 (cmake_policy):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.13")  
-- Performing Test HAVE_LD_VERSION_SCRIPT
-- Performing Test HAVE_LD_VERSION_SCRIPT - Success
-- Symbol prefix: 
-- folder img: converting background.bmp to C file to include in binary
-- Configuring done (1.7s)
-- Generating done (0.1s)
-- Build files have been written to: /home/brixnet/Documents/my-documents/dev/PineTime/InfiniSim/build
```

</details>

[... 196 lines cut ...]
[ 39%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/XPT2046.c.o
[ 39%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/evdev.c.o
[ 39%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/keyboard.c.o
[ 40%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/mouse.c.o
[ 40%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/libinput.c.o
[ 40%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/mousewheel.c.o
[ 41%] Building C object CMakeFiles/sim-base.dir/lv_drivers/win32drv/win32drv.c.o
[ 41%] Building C object CMakeFiles/sim-base.dir/lv_drivers/win_drv.c.o
[ 41%] Building CXX object CMakeFiles/sim-base.dir/sim/FreeRTOS.cpp.o
[ 41%] Building CXX object CMakeFiles/sim-base.dir/sim/task.cpp.o
[ 42%] Building CXX object CMakeFiles/sim-base.dir/sim/timers.cpp.o
[ 42%] Building CXX object CMakeFiles/sim-base.dir/sim/queue.cpp.o
[ 42%] Building CXX object CMakeFiles/sim-base.dir/sim/portmacro_cmsis.cpp.o
[ 42%] Building CXX object CMakeFiles/sim-base.dir/sim/libraries/delay/nrf_delay.cpp.o
[ 43%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/hal/nrf_gpio.cpp.o
[ 43%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/hal/nrf_rtc.cpp.o
[ 43%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/mdk/nrf52.cpp.o
[ 45%] Linking CXX static library libsim-base.a
[ 45%] Built target sim-base
gmake: *** [Makefile:136: all] Error 2
```

</details>

Thanks for reading

FintasticMan, 2023-09-10:
That deprecation warning isn't the source of the error, that's just an outdated minimum cmake version from libpng. The issue is that you don't have the `patch` program installed. That can be installed with the `patch` package.

getlogname, 2023-09-10:
Thanks, I installed `patch` wich resolved my initial problem, but I now have an issue with `swc`:
 <details>
  <summary>Failed build</summary>

```
[  0%] Generating sim_background.h
[  0%] Building C object CMakeFiles/QCBOR.dir/InfiniTime/src/libs/QCBOR/src/ieee754.c.o
[  0%] Building C object CMakeFiles/littlefs.dir/InfiniTime/src/libs/littlefs/lfs_util.c.o
[  1%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_disp.c.o
[  1%] Building C object CMakeFiles/QCBOR.dir/InfiniTime/src/libs/QCBOR/src/qcbor_decode.c.o
[  1%] Building C object CMakeFiles/littlefs.dir/InfiniTime/src/libs/littlefs/lfs.c.o
[  1%] Built target infinisim_img_background
[  1%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_group.c.o
[  1%] Generate fonts and images for resource package
[  1%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_indev.c.o
[  2%] Building C object CMakeFiles/QCBOR.dir/InfiniTime/src/libs/QCBOR/src/qcbor_encode.c.o
[  3%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_obj.c.o
[  3%] Building C object CMakeFiles/QCBOR.dir/InfiniTime/src/libs/QCBOR/src/qcbor_err_to_str.c.o
[  3%] Building C object CMakeFiles/QCBOR.dir/InfiniTime/src/libs/QCBOR/src/UsefulBuf.c.o
[  3%] Linking C static library liblittlefs.a
[  3%] Linking C static library libQCBOR.a
[  3%] Built target littlefs
[  3%] Generating fontawesome_weathericons.c
[  3%] Built target QCBOR
[  3%] Generating jetbrains_mono_42.c
[  3%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_refr.c.o
[  3%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_core/lv_style.c.o
[  3%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_arc.c.o
[  3%] Built target infinitime_fonts_fontawesome_weathericons
[  4%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_blend.c.o
[  4%] Generating jetbrains_mono_76.c
[  4%] Built target infinitime_fonts_jetbrains_mono_42
[  5%] Generating jetbrains_mono_bold_20.c
[  5%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_img.c.o
[  5%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_label.c.o
[  5%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_line.c.o
[  6%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_mask.c.o
[  6%] Built target infinitime_fonts_jetbrains_mono_76
[  6%] Generating jetbrains_mono_extrabold_compressed.c
[  6%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_rect.c.o
[  6%] Built target infinitime_fonts_jetbrains_mono_bold_20
[  6%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_draw_triangle.c.o
[  6%] Generating lv_font_navi_80.c
[  6%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_buf.c.o
[  7%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_cache.c.o
[  7%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_draw/lv_img_decoder.c.o
[  7%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font.c.o
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_dejavu_16_persian_hebrew.c.o
[  8%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  8%] Generating lv_font_sys_48.c
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_fmt_txt.c.o
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_loader.c.o
[  8%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_10.c.o
[  9%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_12.c.o
[  9%] Built target infinitime_fonts_lv_font_navi_80
[ 10%] Generating open_sans_light.c
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_12_subpx.c.o
[ 10%] Built target infinitime_fonts_lv_font_sys_48
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_14.c.o
[ 10%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_16.c.o
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_18.c.o
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_20.c.o
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_22.c.o
[ 12%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_24.c.o
[ 13%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_26.c.o
[ 13%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_28.c.o
[ 13%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_28_compressed.c.o
[ 14%] Building C object CMakeFiles/sim-base.dir/InfiniTime/src/libs/lvgl/src/lv_font/lv_font_montserrat_30.c.o
[... cut ...]
[ 47%] Building CXX object CMakeFiles/sim-base.dir/sim/portmacro_cmsis.cpp.o
[ 47%] Building CXX object CMakeFiles/sim-base.dir/sim/libraries/delay/nrf_delay.cpp.o
[ 48%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/hal/nrf_gpio.cpp.o
[ 48%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/hal/nrf_rtc.cpp.o
[ 48%] Building CXX object CMakeFiles/sim-base.dir/sim/nrfx/mdk/nrf52.cpp.o
[ 49%] Linking CXX static library libsim-base.a
[ 49%] Built target sim-base
gmake: *** [Makefile:136: all] Error 2
```
</details>

NeroBurner, 2023-09-11:
You can try the `libpng_system_lib` branch. On there I tried to build with the `fedora:latest` docker image (I think that is version 35, not sure though). 

I've updated the readme on that branch as well

https://github.com/InfiniTimeOrg/InfiniSim/tree/libpng_system_lib

```sh
sudo dnf install cmake SDL2-devel g++ npm patch perl libpng-devel
```

getlogname, 2023-09-11:
I'm still getting the `swc` errors, tried on both latest Fedora 38 and fresh Fedora 35 VM. However, it works with `cmake -S . -B build -DBUILD_RESOURCES=OFF`.

I'm trying to set-up a safe dev environment to write watch faces and applications without bricking my sealed PineTime, but for this I need `-DBUILD_RESOURCES=ON` , right?

NeroBurner, 2023-09-11:
Only if you want to write apps which use resources from the littles filesystem. If you write apps just part of the regular firmware you can build without the resources generation. So yes you could develop without the resources.

I tested the fedora image last week. Maybe something in the lv_img_conv tool did change since then? If so this should be addressed as well 🤔 (generally I mean)

getlogname, 2023-09-11:
OK thanks for your help, I'll check again later if the build passes with the default parameters before closing this issue.

FintasticMan, 2023-09-12:
Some other people have reported issues like this, where `@swc/core` has received an update that breaks the build. You can apparently fix it by pinning it to an older version.

getlogname, 2023-09-12:
You're right, I managed to build using swc v1.3.80.
Following the main branch readme steps with `npm install ts-node@10.9.1 @swc/core@1.3.80 lv_img_conv@0.3.0` works on latest Fedora.

FintasticMan, 2023-09-12:
Does it fail with 1.3.81? If so, we can try to figure out what changed in that version that causes the issue.

getlogname, 2023-09-12:
I just picked a random version, my bad. It seems **1.3.83** introduced this bug.

FintasticMan, 2023-09-12:
Thanks for figuring that out! Then I think it's likely one of the following commits that introduced the issue: [ee75756](https://github.com/swc-project/swc/commit/ee7575695de6dad140457ffb8bb8f0ac80c4dcdc), [0db25a2](https://github.com/swc-project/swc/commit/0db25a252cf35e4b64b38bde9f34a2f33eb2f662) or [a86e9f3](https://github.com/swc-project/swc/commit/a86e9f3bb5bd490ebf0b18fe7349a2b0fbc0c45f).

FintasticMan, 2023-10-28:
Fixed by InfiniTimeOrg/InfiniTime#1863. @NeroBurner could you close this issue?

closed 2023-10-28 by getlogname