InfiniTimeOrg/InfiniSim #121: Build fails at targetlittlefs-do

coffandro, 2023-09-17:
I run `cmake -S . -B build` which builds fine but when i run `cmake --build build -j4` i get the following output and errors

```
[  1%] Built target infinitime_fonts_jetbrains_mono_42
[  1%] Built target infinitime_fonts_fontawesome_weathericons
[  1%] Built target infinitime_fonts_jetbrains_mono_76
[  2%] Built target infinitime_fonts_jetbrains_mono_extrabold_compressed
[  2%] Built target infinitime_fonts_jetbrains_mono_bold_20
[  3%] Built target infinitime_fonts_lv_font_navi_80
[  3%] Built target infinitime_fonts_open_sans_light
[  3%] Built target infinitime_fonts_lv_font_sys_48
[  3%] Generate fonts and images for resource package
[  4%] Built target littlefs
[  4%] Built target infinisim_img_background
[  6%] Built target QCBOR
[  7%] Built target miniz
[ 10%] Built target infinitime_fonts
[ 62%] Built target sim-base
[ 63%] Building CXX object CMakeFiles/littlefs-do.dir/littlefs-do-main.cpp.o
[ 63%] Building CXX object CMakeFiles/littlefs-do.dir/InfiniTime/src/components/fs/FS.cpp.o
[ 63%] Building CXX object CMakeFiles/littlefs-do.dir/sim/drivers/SpiNorFlash.cpp.o
/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/transpilers/swc.ts:262
      throw new Error(
            ^
Error: @swc/core threw an error when attempting to validate swc compiler options.
You may be using an old version of swc which does not support the options used by ts-node.
Try upgrading to the latest version of swc.
Error message from swc:
Failed to deserialize buffer as swc::config::Options
JSON: {"sourceMaps":true,"module":{"noInterop":false,"type":"commonjs","strictMode":true,"ignoreDynamic":false},"swcrc":false,"jsc":{"parser":{"syntax":"typescript","tsx":false,"dynamicImport":true,"importAssertions":true},"target":"es5","transform":{"legacyDecorator":true,"react":{"throwIfNamespace":false,"useBuiltins":false,"runtime":"automatic"}},"keepClassNames":false,"experimental":{"keepImportAssertions":true}}}

Caused by:
    unknown field `keepImportAssertions`, expected one of `plugins`, `keepImportAttributes`, `emitAssertForImportAttributes`, `cacheRoot`, `disableBuiltinTransformsForInternalTesting` at line 1 column 415
    at createVariant (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/transpilers/swc.ts:262:13)
    at createSwcOptions (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/transpilers/swc.ts:211:25)
    at create (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/transpilers/swc.ts:56:41)
    at createTranspiler (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/index.ts:784:16)
    at createTranspileOnlyGetOutputFunction (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/index.ts:1341:28)
    at createFromPreloadedConfig (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/index.ts:1404:34)
    at phase4 (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/bin.ts:543:44)
    at bootstrap (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/bin.ts:95:10)
    at main (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/bin.ts:55:10)
    at Object.<anonymous> (/mnt/c/Users/Cornelius/infinisim/node_modules/ts-node/src/bin.ts:800:3)
Traceback (most recent call last):
  File "/mnt/c/Users/Cornelius/infinisim/InfiniTime/src/resources/generate-img.py", line 56, in <module>
    main()
  File "/mnt/c/Users/Cornelius/infinisim/InfiniTime/src/resources/generate-img.py", line 51, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/mnt/c/Users/Cornelius/infinisim/node_modules/.bin/lv_img_conv', '/mnt/c/Users/Cornelius/infinisim/InfiniTime/src/resources/images/pine_logo.png', '--force', '--output-file', 'pine_small.bin', '--color-format', 'CF_TRUE_COLOR_ALPHA', '--output-format', 'bin', '--binary-format', 'ARGB8565_RBSWAP']' returned non-zero exit status 1.
gmake[2]: *** [resources/CMakeFiles/GenerateResources.dir/build.make:73: resources/CMakeFiles/GenerateResources] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:567: resources/CMakeFiles/GenerateResources.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[ 64%] Building CXX object CMakeFiles/littlefs-do.dir/InfiniTime/src/drivers/Spi.cpp.o
[ 64%] Building CXX object CMakeFiles/littlefs-do.dir/sim/drivers/SpiMaster.cpp.o
[ 65%] Building CXX object CMakeFiles/littlefs-do.dir/InfiniTime/src/components/settings/Settings.cpp.o
[ 65%] Linking CXX executable littlefs-do
[ 65%] Built target littlefs-do
gmake: *** [Makefile:136: all] Error 2
```

NeroBurner, 2023-09-17:
It's still en error with the `lv_img_conf` npm executable. Seems nothing concerning the `littlefs-do` executable. I think the build will succeed if you disable resource generation

```
Traceback (most recent call last):
  File "/mnt/c/Users/Cornelius/InfiniSim/InfiniTime/src/resources/generate-img.py", line 56, in <module>
    main()
  File "/mnt/c/Users/Cornelius/InfiniSim/InfiniTime/src/resources/generate-img.py", line 51, in main
    subprocess.check_call(line)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/mnt/c/Users/Cornelius/InfiniSim/node_modules/.bin/lv_img_conv', '/mnt/c/Users/Cornelius/InfiniSim/InfiniTime/src/resources/images/pine_logo.png', '--force', '--output-file', 'pine_small.bin', '--color-format', 'CF_TRUE_COLOR_ALPHA', '--output-format', 'bin', '--binary-format', 'ARGB8565_RBSWAP']' returned non-zero exit status 1.
```

Try to run the failing command (the one with `lv_img_conv`) directly in the shell.

NeroBurner, 2023-09-17:
just merged https://github.com/InfiniTimeOrg/InfiniSim/pull/122 stating a minimum version of nodejs of 14.0 and issueing a warning if a lower version is deteted.

Although it seems 14.0 isn't quite the lowest version as our Ubuntu 22.04 base build fails as well

Maybe https://github.com/InfiniTimeOrg/InfiniSim/issues/120 helps you, as they've found that downgrading (or pinning) of the `@swc/core@1.3.80` package (anything below `1.3.83`)  resolves the issue

coffandro, 2023-09-18:
That fixes it

closed 2023-09-18 by coffandro

NeroBurner, 2023-09-18:
The swc thingy or the disable resources flag?

coffandro, 2023-09-20:
Installing the `@swc/core@1.3.80` instead of newest version, i think it should be added to the install instructions

FintasticMan, 2023-09-21:
We are working on a solution that likely won't require downgrading packages.

NeroBurner, 2023-09-22:
Yes Feedback on https://github.com/InfiniTimeOrg/InfiniTime/pull/1863 would be highly appreciated

boldandbusted, 2023-09-28:
Just FYI, this also fixed the same problem for me. Changed the install step to:

```
npm install ts-node@10.9.1 @swc/core@1.3.80 lv_img_conv@0.3.0
```

Cheers.