InfiniTimeOrg/InfiniSim #144: build on MacOS Big Slur fails.

luto65, 2024-06-12:
dear all,
cmake --build build -j4
[ 11%] Building C object CMakeFiles/sim-base.dir/lv_drivers/indev/mouse.c.o
In file included from /Users/bingo/InfiniSim/lv_drivers/indev/keyboard.c:9:
/Users/bingo/InfiniSim/lv_drivers/indev/keyboard.h:36:10: fatal error: 'SDL2/SDL.h' file not found
#include MONITOR_SDL_INCLUDE_PATH

however SDL2 are properly installed and the following test works fine
g++ main.cpp -o main -I/Library/Frameworks/SDL2.framework/Headers -F/Library/Frameworks -framework SDL2

#include <iostream>
#include <SDL.h>
int main(int argc, char **argv){
    if (SDL_Init(SDL_INIT_EVERYTHING) != 0){
        std::cout << "SDL_Init Error: " << SDL_GetError() << std::endl;
        return 1;
    }
    SDL_Quit();
    return 0;
}

Thanks.

luto65, 2024-06-12:
short update:
1. use Xcode instead of unix
cmake -S . -B build -G Xcode
2. append the following line to the clang compilation
-I/Library/Frameworks/SDL2.framework/Headers -F/Library/Frameworks 

however, i have not yet found out how to append that to all compilations...

luto65, 2024-06-14:
thankyou @NeroBurner now i get another error:

Undefined symbols for architecture x86_64:
  "__lv_anim_ll", referenced from:
      __lv_anim_core_init in libsim-base.a(lv_anim.o)
      _anim_task in libsim-base.a(lv_anim.o)
      _anim_mark_list_change in libsim-base.a(lv_anim.o)
      _lv_anim_start in libsim-base.a(lv_anim.o)
      _lv_anim_del in libsim-base.a(lv_anim.o)
      _lv_anim_del_all in libsim-base.a(lv_anim.o)
      _lv_anim_get in libsim-base.a(lv_anim.o)
      ...

NeroBurner, 2024-06-15:
looks to me like not-exported symbols. Does mac need to mark headers as exported like MSVC does?

maybe https://github.com/InfiniTimeOrg/InfiniSim/issues/25 helps?

NeroBurner, 2024-06-15:
This one has the same problem: https://github.com/InfiniTimeOrg/InfiniSim/issues/127

is it OK to close this one as duplicate?

luto65, 2024-06-16:
dear all, i close this build since at the end the solution has been to 
1. download and install https://download.virtualbox.org/virtualbox/7.0.18/VirtualBox-7.0.18-162988-OSX.dmg
2. download OpenSuse tumbleweed https://download.opensuse.org/tumbleweed/iso/openSUSE-Tumbleweed-DVD-x86_64-Current.iso
3. create a new VM using VirtualBox and OpenSuse Tumbleweed https://www.youtube.com/watch?v=HgTSK48hlds
4. start the new VM and then follow the installation steps as described therein.

closed 2024-06-16 by luto65