InfiniTimeOrg/InfiniSim #147: Sim builds successfully (with warnings though) but app crashes on startup

Haui1112, 2024-09-08:
Hi there,
pretty similar issue than #99. Running on pop_os, for me the build completes but the app fails to start. It's possible this stems from me trying to build a couple of times. Not sure if I need to clean the environment beforehand and how to do it.

## Log
Here's the building process in detail as a paste: https://paste.giftedmc.com/?ece837b6adad2321#F8t12Y2raQMKr6WneSNCBsAZkNvFDsczgxbowpm5W2ik

## Issue
The program fails with ```src/libs/littlefs/lfs.c:1070:error: Corrupted dir pair at {0x0, 0x1}```

## Steps to reproduce
1. install pop_os
2. git clone --recursive https://github.com/InfiniTimeOrg/InfiniSim.git
3. cd InfiniSim
4. sudo apt install -y cmake libsdl2-dev g++ npm libpng-dev
5. npm install lv_font_conv@1.5.2
6. python3 -m venv .venv
7. source .venv/bin/activate
8. pip install wheel Pillow
9. cmake -S . -B build
10. cmake --build build -j4
11. ./build/infinisim
12. see error

## Expected behavior
App starts without the error and stays open

NeroBurner, 2024-09-08:
Warnung in build log:
```
CMake Warning at CMakeLists.txt:281 (message):
  Node version v12.22.9

   is less than required 14+, you will probably encounter build errors
```

Haui1112, 2024-09-10:
Thank you very much. I tried on a different machine today and it works well. Will update Node accordingly. Have a good one.

closed 2024-09-10 by Haui1112