InfiniTimeOrg/InfiniSim #149: The `InfiniTime` submodule needs updating.

shymega, 2024-09-11:
For #148 to work correctly, we need to bump the `InfiniTime` submodule to v1.14.1.

This will allow the new Python `lv_img_conv` executable to be found.

Thanks.

NeroBurner, 2024-09-16:
When I update the submodule to the latest release `1.14.0` or `1.14.1` the `InfiniSim` build fails with the following error message

```
[ 98%] Building CXX object CMakeFiles/infinisim.dir/InfiniTime/src/touchhandler/TouchHandler.cpp.o
/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/drivers/Spi.cpp: In member function ‘bool Pinetime::Drivers::Spi::Write(const uint8_t*, size_t)’:
/home/nero/repos/pinetime/InfiniSim/InfiniTime/src/drivers/Spi.cpp:13:25: error: no matching function for call to ‘Pinetime::Drivers::SpiMaster::Write(uint8_t&, const uint8_t*&, size_t&)’
   13 |   return spiMaster.Write(pinCsn, data, size);
      |          ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from /home/nero/repos/pinetime/InfiniSim/InfiniTime/src/drivers/Spi.h:4,
                 from /home/nero/repos/pinetime/InfiniSim/InfiniTime/src/drivers/Spi.cpp:1:
/home/nero/repos/pinetime/InfiniSim/sim/drivers/SpiMaster.h:34:12: note: candidate: ‘bool Pinetime::Drivers::SpiMaster::Write(uint8_t, const uint8_t*, size_t, const std::function<void()>&)’
   34 |       bool Write(uint8_t pinCsn, const uint8_t* data, size_t size, const std::function<void()>& preTransactionHook);
      |            ^~~~~
/home/nero/repos/pinetime/InfiniSim/sim/drivers/SpiMaster.h:34:12: note:   candidate expects 4 arguments, 3 provided
```

do you have an idea which PR introduced the change or what I have to change to fix the InfiniSim build?

NeroBurner, 2024-09-16:
Seems like there are already fixes for other PRs from after 1.14.0 in InfiniSim. Merged https://github.com/InfiniTimeOrg/InfiniSim/pull/145 which fixes the current `main` branch (what is going to be InfiniTime 1.15.0) with fixes for the Always On Display feature

So I can't update to `1.14.1` as requested, but the `main` branch now builds. So I'm closing this as fixed

closed 2024-09-16 by NeroBurner

shymega, 2024-09-16:
I guess I can adjust the PR to build on v1.14.0.

Can you give me a prod once the submodule is bumped to v1.15.0, so I can update the Flake?

NeroBurner, 2024-09-16:
I found a commit in `InfiniSim` to branch off of, that works with `1.14.1` and `1.14.0`: 5ad8c0356e830d9290cad41507e573a7174fa1e5

I've created branch [1.14.x](https://github.com/InfiniTimeOrg/InfiniSim/tree/1.14.x) and the tags for the latest two releases

- https://github.com/InfiniTimeOrg/InfiniSim/releases/tag/v1.14.0
- https://github.com/InfiniTimeOrg/InfiniSim/releases/tag/v1.14.1

I hope you can work with those in the meantime?

shymega, 2024-09-16:
Yes, that should be fine. The only issue I'm having currently is that a) you have to pass a query string to the Flake and b) the Python script seems to fail.

You can reproduce, if you have Nix installed, by running `nix build ".?submodules=1" in the PR branch. I need to investigate further.