InfiniTimeOrg/InfiniSim #158: Watchface Infinineat and Casio G7710 greyed out - fonts issues?

metaphys, 2024-09-29:
I can’t seem to test those watchfaces on the simulator. They are greyed out. Since they are the 2 watchfaces with custom fonts I’m suspectiting this is the issue. I’m trying to reuse the fonts from Casio G7710 but it doesn’t seem to work either. Is there something specific to do in order to have the fonts available in infinisim aside from `npm install lv_font_conv@1.5.2`?

NeroBurner, 2024-09-29:
you'll need to load the ressources.zip into the spi-raw image by running `littlefs-do` helper executable in the same directory as you run `infinisim`

https://github.com/InfiniTimeOrg/InfiniSim?tab=readme-ov-file#littlefs-do-helper

metaphys, 2024-09-29:
Thank you for your answer, where can I find the ressources.zip file?

metaphys, 2024-09-29:
so I found a build/resources/infinitime-resources-1.14.0.zip file
I loaded it with `build/littlefs-do res load build/resources/infinitime-resources-1.14.0.zip`
but now I have a segfault on launching infinisim. Am I still missing something?

metaphys, 2024-09-29:
ok starting from a fresh InfiniTIme dir, there is no more segfault so the problem is from my code now.
Thank you for your help on the infinitime-resources-1.14.0.zip file. Maybe the documentation could be improve to indicate the location and name of the file.

NeroBurner, 2024-09-30:
PR welcome ;)

closed 2024-09-30 by NeroBurner